OpenPHDGuiding/phd2 #443: Calibration sanity check should detect no-movement in east or south

bwdev01, 2015-06-28:
Even though the east/south moves have no bearing on the rate and angle calculations, we should flag situations where the star basically hasn't moved at all.  We have at least one incident in the field where this was a problem.

d33psky, 2017-03-03:
I've had 'star did not move enough' messages -> can we close this ticket ?

bwdev01, 2017-03-03:
No, that is something entirely different.  Please leave the ticket open.
 
  _____  

no-movement in east or south (#443)
 

it on GitHub, or mute
-o1kAhnks5riBtegaJpZM4FNfVp>  the thread.
<https://github.com/notifications/beacon/AMZ5VwiWBcGJLoBn7fRfnbIo1Kc8kLE_ks5
riBtegaJpZM4FNfVp.gif>

bwdev01, 2018-05-22:
Implemented via commit 0c75aab

closed 2018-05-22 by bwdev01