OpenPHDGuiding/phd2 #446: ASCOM driver for QHY5 camera returns mangled sub-frames

bwdev01, 2015-07-23:
User reports that using sub-frames with the ASCOM camera connection for the QHY5 results in mangled sub-frames.  Full frames work correctly.  Extensive debug information is available on the related OpenPHD2 forum thread from Giovanni.

agalasso, 2015-07-24:
we probably cannot do anything about this in phd2 since it appears to be a problem with the QHY ASCOM driver. subframes work fine in PHD2 with all the other ASCOM drivers we know of.

It should be very simple to write a script that retrieves a full frame and a subframe from the camera to demonstrate the problem is independent from phd2.

bwdev01, 2016-01-29:
Problem not reproducible, no further comment from OP in 7 months.

closed 2016-01-29 by bwdev01