OpenPHDGuiding/phd2 #456: Request to display FWHM in larger font for focusing

agalasso, 2015-08-24:
from Stark Labs yahoo group:

This is a request.  Could the FWHM text be shown in large text?  When focusing my guidescope I would like to be able to read the FWHM value shown on the star profile window.

Thanks,

Jeff

hholi, 2015-10-29:
I am looking into making the FWHM more visible for the focusing task.

bwdev01, 2015-10-29:
I wonder if we should be computing HFD here because it might be more stable for the purpose of focusing.  As I recall, the current implementation doesn't do a fit or any smoothing so it might be pretty sensitive to changes in peak brightness.  I've always assumed that's why many of the focusing apps use HFD.

hholi, 2015-10-29:
Yes, good point. 
Shall we use the same view and add HFD as another value to show? As it currently is possible to cycle through Mid Row, Avg Col and Avg Row FWHM.
Would it make sense to add HFD to the values to cycle through and make it the default when going to focus mode?
Or shall we add focusing using HFD as a separate new view?

agalasso, 2015-10-29:
I don't think adding more options to cycle though would help much.  I would be very surprised if many users even know they could cycle through at all.
I think the main issue with the current implementation is that it computes FWHM along a single line.  I believe it was done this way for expediency--very simple and fast to calculate.  I think a better approach would be to do a true HFR (or FWHM) for the entire star and remove the cycling altogether.  I have some code to compute hfr if you want it... send me an email.

agalasso, 2015-10-29:
Simpler will be better...  just a single, meaningful metric (HFD) will be most useful I think.

PR #477 (2015-10-29): Solve #456: Star profile view show FWHM in larger font when the view is higher th…
…an half the systems screen.
Fixes #456 

When the star profile view is larger than half the systems screen the font for FWHM will grow as the height increases.
[image]
![star profile half screen height]
(https://cloud.githubusercontent.com/assets/7082679/10821007/69fa5aa4-7e50-11e5-8cc7-3efbd9b12e0a.png)
[image]
closed 2015-10-29 by agalasso