OpenPHDGuiding/phd2 #458: Request to provide debian packages for Linux

agalasso, 2015-08-26:
specifically, a reguest to have PHD2 available through the Ubuntu package manager

pchev, 2015-08-26:
Andy,
at the moment I have a PHD2 ppa that support this purpose: 
https://launchpad.net/~pch/+archive/ubuntu/phd2

To add phd2 to the Ubuntu package manager you need to follow the steps indicated in "Adding this PPA to your system".

A new version is send for build from my server every time you change PHDVERSION or PHDSUBVER in phd.h.

The build procedure itself is described at the bottom of the BuildingPHD2OnLinux wiki page.

raffienficiaud, 2015-08-26:
Just for your information, cmake is able to generate debian packages but this functionality is right now not fully supported with Launchpad (some environment problem, see http://public.kitware.com/pipermail/cmake-developers/2015-August/026125.html if you are curious).

pchev, 2015-08-27:
Raffi,
Yes I also try to build the deb with cmake, but this is only a small part of the problem. All the Linux distribution have very specific rules to build a package and such a way will never be accepted.

The best solution is to find someone with Debian developer right  to maintain the package directly in Debian using the Debian tools. 
In the mean time the Launchpad ppa is the easiest solution.

agalasso, 2015-08-27:
Patrick, thanks, I have updated the Downloads page on the openphdguiding.org web site to point Ubuntu users to your Launchpad page.
I am closing out this issue. if we have other users looking for pre-built linux packages for other distros we can open a new issue.
Andy

closed 2015-08-27 by agalasso