OpenPHDGuiding/phd2 #461: user reports SBIG camera driver confusing main CCD vs Tracking CCD

agalasso, 2015-09-10:
I just installed my new CCD and found a weird behavior.

During connecting CCD to PHD2 -- it shows an option of selecting "tracking" or "main" CCD.
image: https://goo.gl/IEYH5B

If I select "Yes", it will actually use the main CCD (because I can hear the shuttle noise) but won't get any imaging.
If I select "No", it will then use the tracking CCD and seems working correctly. 

Is this normal or there is something wrong on the driver side?

popovkai, 2015-09-13:
Copy from the PHD google forum

---------------------------------------------

Here is the log file: https://dl.dropboxusercontent.com/u/6544226/PHD2_DebugLog_2015-09-12_232357.txt

When "(Tracking CCD found, use it? No = use main image CCD)" pops up, I first select "YES".
From the log file, it seems getting the correct "size" (FullSize=(648,488)). However, the CCD behavior is not right. I can hearing the shutter sound and no image taken (dark).

Then I tried it again and select "NO" this time.
From the log file, the size seems point to the main CCD (FullSize=(3358,2536)). But it seems getting the right guider image...

agalasso, 2017-10-16:
could use some help from somebody with an SBIG camera