OpenPHDGuiding/phd2 #468: EventMonitoring bug: Set_Lock_Position not respected when starting guiding/find star

opoloko, 2015-10-12:
Hi,

I have a client that controls PHD2 by JSON/RPC, and I discovered that once I set the lock position correctly with 'set_lock_position', and I verify it's correctly setup with 'get_lock_position' (is in a correct position, working on simulator), when I start 'guide' (or use undocumented 'find_star' method) then PHD ignores completely my previously set lock position and performs anyway an AutoFindStar.

Is there any way you can fix this so I can set a lock position and have the guide (trying at least!) finding the star in that position and not performing always and anyway AutoFindStar? Otherwise I don't see the point of set_lock_position remote command.

Thanks!

opoloko, 2015-10-12:
Looking at event_server source it seems indeed you perform an automatic star select, while it should really just select star if a given position is previously set:

static void find_star(JObj& response, const json_value *params)
{
    VERIFY_GUIDER(response);

    bool error = pFrame->pGuider->AutoSelect();

    if (!error)
    {
        const PHD_Point& lockPos = pFrame->pGuider->LockPosition();
        if (lockPos.IsValid())
        {
            response << jrpc_result(lockPos);
            return;
        }
    }

    response << jrpc_error(1, "could not find star");
}

agalasso, 2015-10-12:
opoloko,

find_star *is* the Auto-select function. It was not meant to be undocumented; its omission from the wiki was an oversight and that has now been corrected.

we can investigate the issue with set_lock_position not having the desired effect.

It would help if we could understand what you are trying to accomplish though.  I am interested to know why you do not want to just use the guide method to start guiding?

Andy

opoloko, 2015-10-12:
Hi Andy,

simply my app stores the last known locked guiding star (that might be user selected and not automatic selected), and in case of certain events/problems, it stops/restarts looping and guiding of PHD.

Problem is that I want to be able to restart the guiding using coordinates of MY previous guide star, not the automatic one that PHD locks when restarting the guide via RPC.

There are many reason to the fact that I might not want to use automatic star select, the basic problem is just that if I set a lock star position then when I start guiding, PHD should use that position not another one.

Thanks for your help and feedback :)

agalasso, 2015-10-12:
opoloko,

Thanks, as I said, we will investigate and address the problem with set_lock_position.

The reason I was asking about why you did not want to use the guide method was probing whether there is was an actual problem you experienced with auto star selection. Several of us (myself included) rely on auto-star selection for automated imaging (via SGP or other apps) and if there was some shortcoming of auto star selection I wanted to know about it since improving it could benefit many users.

Andy

agalasso, 2015-10-13:
opoloko,

Here is the sequence  of server commands that will accomplish what you were asking for:

1. start looping
{"method":"loop","id":"whatever"}
2. set the lock position, with the parameter exact = false. This will select the star near the specified coordinates. Be sure to check the return status as that will indicate whether or not the star was selected. This is the server command equivalent of clicking on the star in the UI.
{"method":"set_lock_position","params":[615,390,false],"id":1}
3. use the guide method to start guiding
{"method": "guide", "params": [{"pixels": 1.0, "time": 8, "timeout": 100}, false], "id": 42}

If you have further questions or issues, please post a message in the open phd guiding google group.  It's better to discuss development issues in that forum rather than through the issue ticket system.

Andy

closed 2015-10-13 by agalasso