OpenPHDGuiding/phd2 #478: Norwegian translation

hholi, 2015-10-29:
Add norwegian translation.

agalasso, 2015-10-29:
in case you hadn't seen it:
https://github.com/OpenPHDGuiding/phd2/wiki/TranslatingPHD2

hholi, 2015-10-29:
Thanks, I saw that. When editing PHD settings it seems that it is only possible to select Default and English. I expected the drop-down to contain all languages in the locale folder. Is it something that I am missing?

agalasso, 2015-10-29:
I believe you only see Default and English if you are running from the debugger.  You can put messages.mo and messages.po under the locales directory of a phd2 installation and you will see the new language when you run the installed phd2.

d33psky, 2017-03-03:
Is this done ? Can this ticket be closed ?

agalasso, 2017-03-03:
this is done!

closed 2017-03-03 by agalasso