OpenPHDGuiding/phd2 #485: PHD2 instance 1 is already running.

d33psky, 2015-11-18:
This code piece in phd.cpp checks for a lock file : 

```
106     m_instanceChecker = new wxSingleInstanceChecker(wxString::Format("%s.%ld", GetAppName(), m_instanceNumber    ));
107     if (m_instanceChecker->IsAnotherRunning())
108     {
109         wxLogError(wxString::Format(_("PHD2 instance %ld is already running. Use the "
110             "-i INSTANCE_NUM command-line option to start a different instance."), m_instanceNumber));
```

on Linux named "phd2.1" in the users' home directory. It does not detect stale lock files :(
(where no other phd is running.)
This is related to bug http://trac.wxwidgets.org/ticket/17046

A workaround is to ```rm phd2.1```. We need something better.

/usr/bin/wx-config --version-full : 3.0.2.0
git branch master of today. basically 2.5.0dev7
OS: Ubuntu

EverettQuebral, 2016-01-29:
I have the same problem with the latest Mac OSX version 10.11(El Capitan).  To get around with this, I compiled everything with the latests XCode and OS X 10.11

More information here http://everettquebral.com/2015/10/17/phd-2-5-instance-already-running-error/ and installer available for download.

pchev, 2016-03-04:
I cannot reproduce this problem. 
Looking at wx/src/unix/snglinst.cpp it include a check for the running process using kill(m_pidLocker, 0) and delete the lock file if not. This exist in every version of the file.

Running phd2 on Ubuntu 14.04 with wx 3.0.0  I receive a message "Deleted stale lock file.." 
On Ubuntu 15.10 with wx 3.0.2 the message is suppressed and the stale lock file is replaced silently. This correspond to this patch that suppressed the message: http://trac.wxwidgets.org/ticket/15944

17046 is a proposition to replace the lock file by a shared memory, this is not related to this problem.

You must try to add a breakpoint and run step by step in snglinst.cpp to find why your lock file is not deleted.

EverettQuebral, 2016-03-08:
The problem is reported using Mac OS X 10.11, a recompilation should fix the issue.

d33psky, 2017-03-03:
I'm happy to report that I have not seen this issue for months. (still on ubuntu 14.04 with wx 3.0.2.0).

agalasso, 2017-03-03:
let's close it to get it off the pile.  if we see it again we can open a new ticket or reopen this one.

closed 2017-03-03 by agalasso