OpenPHDGuiding/phd2 #496: Support for Orion SSAG

ClundXIII, 2016-01-09:
Hi,

recently my gf bought the Orion StarShoot AutoGuider
According to some googling, this is a QHY5 device. I found an udev rule from this thread:
http://www.cloudynights.com/topic/459674-iastrohub-tablet-astrophotography-comes-true/?p=6126387

the udev rule itself is this one:
https://drive.google.com/file/d/0B3RSuv91xWggaFQzbldPcUVCOUU/edit

Can you include these rule in your ubuntu, etc packages?
Or should I report this missing rule on some other bugtracker?

Also it might be useful to provide a udev rule list for all supported cameras in general.
We can confirm it is working now.

Clund

(Also I had to run "sudo phd2" in order to get it working, I don't know if that has something todo with udev)

pchev, 2016-01-10:
Hi,

The rule file you mention load firmware to the camera from /lib/firmware/ccd/qhy*.hex
This files are not installed by PHD2, so I think it is to the package that install this files to add the udev rule. Do you know from where it come? 

I not use a qhy camera but I have this files and rules on my system installed by oaCapture at another location. It will be very messy if every application install it's own version because you end-up with firmware/driver mismatch.

You must avoid running as root. 
The problem is write access right to the device in /dev because your rule set MODE="0664" and GROUP="video".
To solve the problem just add your user to the group video.

ClundXIII, 2016-01-10:
Well I installed some firmware package, too.
I think this one: http://sourceforge.net/projects/cccd/files/firmware/

phd2 is from launchpad (very up-to-date)
Maybe the firmware solved the problem and I didn't need the rule. (I tried sudo after I already installed both).
But lsusb didn't show any name until I had the udev rule.

d33psky, 2017-03-03:
Not sure if this issue even belongs to PHD2. If we decide it belongs to PHD2 I think it's best placed in the package install files' post scriptlets, but that may clash with distribution udev rules for the same cameras.

ClundXIII, 2017-03-04:
You can close this if you want.

I suggest creating a help file / page with all devices and how to get them working.

agalasso, 2017-03-04:
I have updated the [supported cameras wiki page](https://github.com/OpenPHDGuiding/phd2/wiki/CameraSupport).

closed 2017-03-04 by agalasso