OpenPHDGuiding/phd2 #504: request to determine "reverse dec output after meridian flip" setting automatically

agalasso, 2016-01-31:
Chris Rowland suggested the idea of performing a pair of calibrations, one on each side of the meridian, to automatically determine the  "reverse dec output after meridian flip" setting.
perhaps use a wizard type interface to prompt the user when to do the meridian flip.

bwdev01, 2016-03-04:
Marginal utility, not that difficult a problem for users to figure out.

bwdev01, 2018-09-13:
From Andy:

If we were to do it I was envisioning a button next to the setting itself that would run the wizard.  Maybe we would also integrate it into the new equipment wizard, perhaps a checkbox similar to "build dark library" that says "calibrate meridian flip settings" or something like that (only present when pointing state is available)

agalasso, 2018-10-03:
Fixed by 68b5a6a15ad5f01598496b46261615abc26d680e

closed 2018-10-03 by agalasso