OpenPHDGuiding/phd2 #509: tooltips: clarify meaning of pixel size binned vs unbinned

agalasso, 2016-02-23:
From Vince:

I was confused however with minimum move pixel size, but it seems to me from looking at my logs in PHD Log Viewer and applying the "Limits" graph, that it refers to the binned pixel size, so that for the QHY5L-II binned 2x2, the pixel size used is actually 7.50 um.  Perhaps a note in the minimum move mouse-over info explaining this in a future version would be helpful to others.

agalasso, 2016-02-23:
From Jim:

I did not know what pixel size to set under Camera-specific properties, Camera, Advanced Settings, the actual size or the binned size, which is correct please when using binning?

bwdev01, 2016-03-10:
Fixed - next release after 2.6.1dev5

closed 2016-03-10 by bwdev01