OpenPHDGuiding/phd2 #510: some questions about backlash

ZhuXinyan, 2016-02-24:
I found that there is a state called **CALIBRATION_STATE_CLEAR_BACKLASH** in the switch in the function **Scope::UpdateCalibrationState**。I am really confused that why it can clear backlash espically why the mount should go north like 
**pMount->ScheduleCalibrationMove(this, NORTH, m_calibrationDuration);**
I hope for your help because I am a green hand。
Excuse me for any mistakes since my not-so-good English。

agalasso, 2016-02-24:
Hello ZhuXinyan
we use the issue tracker for tracking bugs and feature requests. We'd be glad to discuss your question in the open-phd-guiding google group, or you can email me directly andy.galasso at gmail dot com

closed 2016-02-24 by agalasso

bwdev01, 2016-02-24:
Well, I$B!G(Bm sorry, but I don$B!G(Bt really understand what you$B!G(Bre asking.  Most
mounts have some amount of declination backlash, and for some mounts it can
be very large.  In the latter case, the declination calibration can be badly
affected unless the backlash is cleared first.  The calibration phase you$B!G(B
re talking about attempts to clear Dec backlash by moving the scope north
until PHD2 sees a consistent pattern of movement in the same direction with
no reversals.  Once that$B!G(Bs accomplished, PHD2 continues to move the scope
north to compute the guide rate.  Why are you asking?  If you$B!G(Bre having
some sort of problem with calibration, that$B!G(Bs what we should be talking
about.

Bruce

  _____

switch in the function Scope::UpdateCalibrationState$B!#(BI am really confused
I hope for your help because I am a green hand$B!#(B
Excuse me for any mistakes since my not-so-good English$B!#(B
-
<https://github.com/OpenPHDGuiding/phd2/issues/510> GitHub.
<https://github.com/notifications/beacon/AMZ5V9fOzFfQpzDj2aQb46VTzw1ziHTZks5
pnTjCgaJpZM4HhXEB.gif>

ZhuXinyan, 2016-02-24:
Thank you very much for your help. I think I might have understood what you said. Thank you again for your help