OpenPHDGuiding/phd2 #519: simulator stops guiding in dec ?

d33psky, 2016-03-20:
Simulator run, seem guiding stops in dec after a while. 

[image]

OS : Linux.

agalasso, 2016-03-21:
could you attach the debug log?

d33psky, 2016-03-22:
[PHD2_DebugLog_2016-03-20_225820.txt](https://github.com/OpenPHDGuiding/phd2/files/183719/PHD2_DebugLog_2016-03-20_225820.txt)

bwdev01, 2016-03-22:
You're doing this with a simulated AO!  What you're seeing is expected
behavior when the AO has reached the limit of its travel and the mount needs
to be bumped.  The Dec corrections from the AO stop until the mount bump
returns the guide star back into a region where the AO can resume.  Since
you don't have a mount connection, bumping is not going to work. 
 
Bruce
 
  _____  

 

<https://github.com/OpenPHDGuiding/phd2/files/183719/PHD2_DebugLog_2016-03-2
0_225820.txt> 
-
it on GitHub
<https://github.com/notifications/beacon/AMZ5V3QE1WtchIMsq3qHN1Yj4xIG2XWCks5
pv6azgaJpZM4H0xsZ.gif>

d33psky, 2016-03-22:
Yeah, the mount simulator seems gone. I can only select INDI,none,on-AO,on-Camera . I selected one of the on- options and got this result. Do we need a new ticket for the mount-simulator on Linux ?

agalasso, 2016-03-22:
actually, we never had a mount simulator per se.  To use the internal phd2 simulators you would use:

Camera = Simulator
Mount = On-camera
AO = AO Simulator    or None

closed 2016-03-22 by agalasso

bwdev01, 2016-03-22:
If you really want to see what's going on, you should use the AO graph when
you're using the AO simulator - that and connect to the "On camera" mount as
Andy said.
 
Bruce
 
  _____  

 

it on GitHub
<https://github.com/notifications/beacon/AMZ5V5iZUBP5WCiT3nZVQYrlI5Cfl9pJks5
pwCUfgaJpZM4H0xsZ.gif>

d33psky, 2016-03-22:
With mount = on-Camera the simulator works well \o/
With mount = on-AO the 'connect' button greys-out. This must have been the setup I intended to test.

I find it confusing that Camera can be set to Simulator while Mount cannot. Especially when an AO is added. To me 'mount = on-camera' means ST-4, and both my camera and my AO have ST-4 that can be connected to the mount.
In Real Life I want to connect with Mount set to an INDI mount interface, not ST-4 at all.

agalasso, 2016-03-23:
I agree it is confusing.
Probably not worth changing at this point unless people feel really strongly about it. We do have a section dedicated to the simulators in the help file (openphdguiding.org/manual/?section=Basic_use.htm#Equipment_Connection)