OpenPHDGuiding/phd2 #520: INDI: detect side of pier

agalasso, 2016-03-22:
can INDI report side of pier?

if so, then we should have phd2 use the pier side information in the same we we do for ASCOM.

d33psky, 2016-03-27:
Created a feature request ticket for INDI at [https://sourceforge.net/p/indi/bugs/72/](https://sourceforge.net/p/indi/bugs/72/)

pchev, 2016-05-02:
I remark that INDI Eqmod driver implement a PIERSIDE property.
I just commit an update for scope_INDI.cpp to return SideOfPier accordingly.
This need some testing because I am not sure if the returned value use the same rules as ASCOM.

pchev, 2016-05-17:
Tested yesterday with indi_eqmod and HEQ5.
I set a 1 degree polar alignment error to make dec correction evident, calibrating with pointing east, then guiding pointing east and west after mount flip. Everything work as expected.

closed 2016-05-17 by pchev