OpenPHDGuiding/phd2 #521: INDI with AO

d33psky, 2016-03-25:
I'm pretty sure we supported AO via INDI.
It's gone now from the options list (AO only has None and Simulator).
2.6.1dev6

agalasso, 2016-03-25:
The only AO we have support for is the SX AO via the built-in driver code in PHD2. We have not implemented any other ao support other than the simulator, and the SX AO only works on windows.
Of course we would like to support the SX AO on other platforms either natively or through INDI, but nobody has ever worked on it.

closed 2016-03-25 by agalasso

d33psky, 2016-03-26:
OK, cool. Guess I have a new project then ;-)

bretm151, 2016-03-26:
It should not be too hard -- the AO support was originally written to be
pretty generic, and the SX AO is a serial port device, which "should be"
easy to talk to.

Bret