OpenPHDGuiding/phd2 #529: Update build to use wx 3.1.0

agalasso, 2016-04-07:
wxWidgets 3.1.0 is now available; let's upgrade

d33psky, 2016-04-11:
I've added wx3.1.0 to the H__buntu Ubuntu 14.04.4 LTS buildslave. It fails, starting with : 

/home/phd2/buildslave/H__buntu/source/calreview_dialog.cpp:138:73: error: ‘wxADJUST_MINSIZE’ was not declared in this scope
     graphLegendGroup->Add(labelRA, 0, wxALIGN_CENTER_VERTICAL | wxALL | wxADJUST_MINSIZE, 5);

Rest of the errors : http://gcc10.fsffrance.org:8011/builders/H__buntu/builds/609/steps/compile/logs/stdio

d33psky, 2016-04-16:
FYI I'm working on this. So removing wxADJUST_MINSIZE and the next are reshuffling of the arguments on loads of SetCellValue

d33psky, 2016-04-16:
Committed. https://github.com/OpenPHDGuiding/phd2/commit/e7bbc8c95c2c0105dd06ef7c5017e77969bc4144
also tested on a wx3.0.2 system.

agalasso, 2016-05-22:
using wxWidgets 3.1.0 starting with 2.6.1dev8

closed 2016-05-22 by agalasso