OpenPHDGuiding/phd2 #532: wxWidgets dialogs unusable due to rendering

d33psky, 2016-04-15:
Two users reported dialog rendering issues on Linux : 
1. Fedora 21 in https://groups.google.com/forum/#!topic/open-phd-guiding/GhW4XB58wdI
2. openSUSE leap 42.1 in https://groups.google.com/forum/#!topic/open-phd-guiding/KsIVwoFedoE

[image]

[image]

d33psky, 2016-08-28:
[image]

domokos, 2016-10-05:
I actually had the same issue on Fedora 22 and made a workaround, which - at least - makes it usable. I changed all the dimensions of the spincontrols to wxDefaultSize (which is  -1 actually). It is not nice, but works.

The whole code is rather ugly in this respect, sizes are just hardcoded like this example in scope.cpp:

 m_pCalibrationDuration = new wxSpinCtrl(GetParentWindow(AD_szCalibrationDuration), wxID_ANY, wxEmptyString, wxPoint(-1, -1),  wxSize(width+30, -1), wxSP_ARROW_KEYS, 0, 10000, 1000,_T("Cal_Dur"));
It seems to be  trying to be relative to "width", but "width"  is just an auto variable set to a fixed value, probably used to set a size, which looked good to who implemented it first.

But elsewhere - like in darks_dialog.cpp it is not even thought of it is just "1" and "20" although the dev bothered to create a - local  - class for wrapping but did not use for sizes.
m_pDarkCount = NewSpinnerInt(this, width, pConfig->Profile.GetInt("/camera/darks_num_frames", DefDarkCount), 1, 20, 1, _("Number of dark frames for each exposure time"));

The correct solution would be to reorganize these to something centralized, which - based on display size, dpi, window size, etc. - would determine the correct values to make it a bit more responsive/portable.

Other sizers suffer from the same issue e.g. the guide statistics' summary on the scope panel - in F22 - where the text is just out of the window.

agalasso, 2016-10-05:
> The correct solution would be to reorganize these to something centralized, which - based on display size, dpi, window size, etc. - would determine the correct values to make it a bit more responsive/portable.

That's a great idea. We would welcome a pull request.

d33psky, 2016-11-25:
Not entirely sure if this is the same issue, but these buttons here unreadable.
[image]
git v2.6.2-61-g4c385f8 , Ubuntu 16.04.1 LTS

domokos, 2016-11-26:
By the nature of the code view not isolated from other parts. There will always be glitches like this. To do it better would be a huge work as sometimes the business logic itself is not distinguished from rendering. 

Try different settings on your desktop (change themes, fonts/sizes, gtk settings, kde rendering of gtk items and so on).

If this does not work fork and fix the parts for your taste in a branch.

Unfortunately there is no easy fix for this issue.

pchev, 2016-11-26:
Sure that separating the view code can help to fix some display problem, even if I not find too difficult to find the right piece in the actual code.

But we cannot fix the code to accommodate every buggy gtk theme, even if this is a distribution default theme.
The reference is to run with the default gtk rendering, without any theme. For that delete any .gtkrc* from your home directory.
In this case there is not a single display problem in PHD2.