OpenPHDGuiding/phd2 #543: documentation and new profile wizard should emphasize BPM

agalasso, 2016-06-22:
it seems like we currently emphasize dark library over bad pixel maps.  It should probably be the other way around

bwdev01, 2016-06-22:
Well, I think we need to be careful to maintain the "P-H-D" philosophy.
Building a dark library is an easy fire-and-forget operation, and for many
cameras - certainly the ones I've used - it works very well.  Building a BPM
is not that automatic because the default settings may not be appropriate.
The UI is more complex and I think it requires more understanding about
what's going on in order to make adjustments and get a good result.  Based
on what I've seen from user logs, we have a great many users who don't do
anything at all in this area, not even a dark library.  The current
approach, essentially the new-profile-wizard, at least pushes them to take
the simple approach.  Finally, none of this makes much difference unless the
user is having recurring problems with hot pixels - a problem that seems to
be all too common with Lodestars but perhaps not many others.  I'd be fine
with beefing up the help doc with regard to hot-pixel problems, but I'm not
sold yet on making UI changes at the expense of complexity.  Just my
2-cents.
 
Bruce 
  _____  

emphasize BPM (#543)
 

<https://github.com/OpenPHDGuiding/phd2/issues/543> , or mute
uoks5qOZKwgaJpZM4I8J-8>  the thread.
<https://github.com/notifications/beacon/AMZ5VyUrSbdmy8ssD42nvEPCRebyPx4Tks5
qOZKwgaJpZM4I8J-8.gif>

agalasso, 2016-06-22:
Thanks Bruce, excellent points. Should we close this issue?

bwdev01, 2016-06-22:
I'd vote in favor of morphing the issue toward having some explicit coverage
in the help doc for hot-pixel problems - something to push the user in the
direction of a bpm when they see those problems.  Right now, there's nothing
much there.
 
 
  _____  

should emphasize BPM (#543)
 

it on GitHub, or mute
fnks5qObNQgaJpZM4I8J-8>  the thread.
<https://github.com/notifications/beacon/AMZ5Vxm_jUXMTSVYNjWie7KSyu42ZnGdks5
qObNQgaJpZM4I8J-8.gif>

bwdev01, 2016-06-23:
Redirected to issue for help doc improvements, issue 544

closed 2016-06-23 by bwdev01