OpenPHDGuiding/phd2 #554: main window size grows 28 pixels in height on new instantiation

d33psky, 2016-08-28:
- start PHD2, use xwininfo to measure window size. In this case : 958x738
- stop PHD2
- start PHD2 again, use xwininfo again, size is now: 958x766
- stop PHD2
- start PHD2 again, size is now 958x794

So the height grows with 28 pixels each time PHD2 starts up.

This has been the case for at least over a year now but the workaround (manual window resize) is so easy that I did not bother making a bug for it. Still it should be fixed as it's annoying.

Tested on Ubuntu 16.04.1, PHD2 v2.6.2-6-g7e7a335

pchev, 2016-08-28:
The problem is not present when using Ubuntu with a KDE or LXDE desktop.
I suspect a side effect of the Ubuntu main menu management as the problematic height correspond to the menu height.
Do you try with and without setting UBUNTU_MENUPROXY ?

d33psky, 2016-08-28:
Also with ```export UBUNTU_MENUPROXY=0``` phd2 grows 28 pixels in height every time.
On another system I can confirm that the issue does now show with KDE. So this may be a gnome/unity issue.