OpenPHDGuiding/phd2 #556: Improve handling of on-the-fly changes to camera binning

bwdev01, 2016-09-04:
If users change binning without also using a separate profile, guiding parameters and performance display tools are not adjusted automatically.  This is confusing to users and generates support issues.  When the binning changes, the following things also need to be done:
1. Performance-related graphs that use the current image scale need to be reset/cleared.
2. Guiding algo parameters should be reset to their default values.
3. Min-move values should be restored to the image-scale dependent settings that are computed by the new-profile-wizard

If a binning change is detected from an outside-of-PHD2 change (e.g. an ASCOM property dialog change), an alert message should be generated in addition to the above changes.

agalasso, 2016-09-04:
> Min-move values should be restored to the image-scale dependent settings that are computed by the new-profile-wizard

I think we can scale the values (e.g., if binning goes from 1 => 2 them min_move goes to min_move/2); there may not be any value from the G.A. if it was never run.

bwdev01, 2016-09-04:
This min-move calculation was centralized as part of the 'reset' feature in
the algo tab.  The calculation in the new-profile-wizard isn't strictly
linear and puts a floor on the value.  We wouldn't want to go from a
min-move of 0.15 to 0.05 for example.  The GA doesn't factor into this, I'm
just referring to an empirical calculation that seems to work pretty well.
If we simply force a 'reset' on the guide algorithm parameters (which we
need to do), the min-move values will already be reset in this way, for
free.  Seems like this is the easiest way to restore things to a known
baseline that ought to work.
 
  _____  

camera binning (#556)
 

it on GitHub, or mute
_SOz5nIks5qmi5NgaJpZM4J0cBX>  the thread.
<https://github.com/notifications/beacon/AMZ5Vw1HaBWrjaxg5Ap9QI-BCKgpaWBnks5
qmi5NgaJpZM4J0cBX.gif>

agalasso, 2016-09-04:
cool, sounds good to me

bwdev01, 2016-09-11:
Completed in 2.6.2dev3

closed 2016-09-11 by bwdev01