OpenPHDGuiding/phd2 #559: Error compiling on rasbian jessie with ASI support

johnwebbcole, 2016-10-05:
I'm trying to compile on a Raspberry Pi 2 for use with my ASI120MM guide camera, but I'm getting the following error:

```
Linking CXX executable phd2
CMakeFiles/phd2.dir/cam_ZWO.cpp.o: In function `Camera_ZWO::Connect(wxString const&)':
cam_ZWO.cpp:(.text+0x4fa0): undefined reference to `ASIInitCamera'
collect2: error: ld returned 1 exit status
CMakeFiles/phd2.dir/build.make:2892: recipe for target 'phd2' failed
make[2]: *** [phd2] Error 1
CMakeFiles/Makefile2:126: recipe for target 'CMakeFiles/phd2.dir/all' failed
make[1]: *** [CMakeFiles/phd2.dir/all] Error 2
Makefile:147: recipe for target 'all' failed
make: *** [all] Error 2
```

I noticed that the `cameras/zwolibs` directory did not have the `armv7` (pi 2) or `armv8` (pi 3) directories.  The ZWO 0.4.092 sdk does have those directories so I copied them into the `zwolibs` directory, but I get the error above.  The sdk has .h files but no .lib files.

Any suggestions to get this to compile?

johnwebbcole, 2016-10-06:
A little more info.  I loaded a fresh raspbian jessie on a Pi3, copied the asi-sdk arm8 files and was able to build PHD2, and it runs.  It cannot connect to my ASI120MM camera and returns the following error in the debug log.

```
11:05:55.269 01.535 1957507632 GetString("/profile/2/cam_hash/524195a3/whichCamera", "") returns ""
11:05:58.534 03.265 1957507632 Status Line: Connecting to Camera ...
11:05:58.537 00.003 1957507632 GetString("/profile/2/cam_hash/524195a3/whichCamera", "") returns ""
11:05:58.537 00.000 1957507632 Connecting to camera id = []
11:05:58.538 00.001 1957507632 ASIOpenCamera ret 5
11:05:58.538 00.000 1957507632 wxMessageBoxProxy(Failed to open ZWO ASI Camera.)
11:06:07.793 09.255 1957507632 Throw from /home/pi/src/phd2/gear_dialog.cpp:970->DoConnectCamera: connect failed
11:06:07.793 00.000 1957507632 Status Line: Camera Connect Failed
```
Any ideas?

I'm going to run the same thing on a fresh image load on a Pi2.

agalasso, 2016-10-06:
From AsiCamera2.h, code 5 is:

> ASI_ERROR_CAMERA_REMOVED, //failed to find the camera, maybe the camera has been removed

since the ZWO SDK is a closed-source binary, I think you'll need to get ZWO involved to diagnose the problem (I'd start by asking on the ZWO Users Forum)

pchev, 2016-10-06:
Do you try to connect your camera using the INDI driver?
This ASIInitCamera missing from some version is strange, but I find it is present for armv7 and armv8 in the SDK  V0.4.0920 available from ZWO. The file you need is libASICamera2.a present in the lib directory.
I cannot help more because I not have a ZWO for testing.

johnwebbcole, 2016-10-07:
I've downloaded the indi drivers and compiled them, but I'm not sure I've gotten them set up correctly.  In the mean time, I found a hardcoded armv6 in the third party cmake file. Changing that to armv7 and recompiling attached to the camera on a Pi 3.

I was able to get the Pi 3 to guide last night, but after that it was loosing the star too much and I switched to a Win laptop.

d33psky, 2017-03-03:
Hey johnwebbcole, do you still want to follow up on this bug ?

pulsar67, 2017-03-19:
Hello, 
I have the same issue while trying to compile phd2 on a Pi3 on Raspbian Jessie.

`[ 97%] Building CXX object CMakeFiles/phd2.dir/wxled.cpp.o
[ 98%] Linking CXX executable phd2
CMakeFiles/phd2.dir/cam_ZWO.cpp.o: In function `Camera_ZWO::Connect(wxString const&)':
cam_ZWO.cpp:(.text+0x4fa0): undefined reference to `ASIInitCamera'
collect2: error: ld returned 1 exit status
CMakeFiles/phd2.dir/build.make:3066: recipe for target 'phd2' failed
make[2]: *** [phd2] Error 1
CMakeFiles/Makefile2:142: recipe for target 'CMakeFiles/phd2.dir/all' failed
make[1]: *** [CMakeFiles/phd2.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
`

I copied the armv7 and armv8 directories from the v0.6.0110 ZWO SDK. 
I understand it's just a link problem but I don't find how to correct it. Any suggestion?

Thank you by advance.

closed 2017-04-26 by agalasso

agalasso, 2018-09-02:
@dmnkhhn, you should be able to build phd2 from scratch on arm8 with no changes to CMakeLists.txt and without downloading anything from ZWO.
What errors do you see when you run a regular build from a clean (unmodified) source tree? https://github.com/OpenPHDGuiding/phd2/wiki/BuildingPHD2OnLinux

commit e3e8459e (2017-04-26): update ZWO Linux/Mac SDK to V0.6.0328; add armv7 and armv8 ASI libs to build - fixes #559