OpenPHDGuiding/phd2 #561: Lots of missing buttons in "Guiding Assistant" for small screen netbooks

cytan299, 2016-10-15:
Hi guys,
   I use a netbook with a small screen to control my mount. Everything works *except* Guiding Assistant because the Start, Stop and Show graph buttons are missing. See picture that was attached to this bug report (phd2_missing_buttons.jpg)

Is it possible to have code to check for screen size, so that the buttons can appear on the side rather than at the bottom, or make the panel bigger.

Thanks.

cytan

[image]

bwdev01, 2016-10-15:
This is problematic, and there are some minimum screen-size requirements for
PHD2.  Perhaps you could try this: http://www.gimespace.com/gms.html
 
And let us know if it works for you.
 
Thanks,
 
Bruce
 
  _____  

Assistant" for small screen netbooks (#561)
 

 
<https://cloud.githubusercontent.com/assets/13826223/19413537/926c93fc-92f5-
11e6-991c-b29f599c78c0.jpg> phd2_missing_buttons
-
<https://github.com/OpenPHDGuiding/phd2/issues/561> , or mute
ihE_wkBks5q0UevgaJpZM4KX0kh>  the thread.
<https://github.com/notifications/beacon/AMZ5V2H6eeK0gGMjZhICzDhKXYhORQqXks5
q0UevgaJpZM4KX0kh.gif>

cytan299, 2016-10-16:
Hi Bruce,   Great suggestion! I just installed gimespace on my netbook and after fiddling with it and specifying that the guide assistant panel has to be 650 pixels high, now I can see the missing buttons.
cytan

   

 
 
 
 

 
 
<https://cloud.githubusercontent.com/assets/13826223/19413537/926c93fc-92f5-
<https://github.com/notifications/beacon/AMZ5V2H6eeK0gGMjZhICzDhKXYhORQqXks5

bwdev01, 2016-10-16:
Thanks for trying it and reporting back, Cytan.  This could be a useful
option for others. 
 
Bruce
 
  _____  

 

<https://cloud.githubusercontent.com/assets/13826223/19413537/926c93fc-92f5-
<https://github.com/notifications/beacon/AMZ5V2H6eeK0gGMjZhICzDhKXYhORQqXks5

it on GitHub, or mute
b2TZLpYks5q0Wx8gaJpZM4KX0kh>  the thread.
<https://github.com/notifications/beacon/AMZ5V8J9FgrSZuoxu-lV-pIPoX6ke3sjks5
q0Wx8gaJpZM4KX0kh.gif>

bwdev01, 2016-10-16:
Hi Cytan.  We're going to take a look at the Guiding Assistant layout and
try to restructure it a bit.  Can you tell me what your screen resolution is
- X pixels wide by Y pixels high?
 
Thanks,
Bruce
 
  _____  

 
 
<https://cloud.githubusercontent.com/assets/13826223/19413537/926c93fc-92f5-
<https://github.com/notifications/beacon/AMZ5V2H6eeK0gGMjZhICzDhKXYhORQqXks5

cytan299, 2016-10-16:
blockquote, div.yahoo_quoted { margin-left: 0 !important; border-left:1px #715FFA solid !important; padding-left:1ex !important; background-color:white !important; }  Hi Bruce,
  It's 1024x600
  I can test it out once you have a fix.
  Thanks for a great piece of software.
Cytan

 
 

 
 
<https://cloud.githubusercontent.com/assets/13826223/19413537/926c93fc-92f5-
<https://github.com/notifications/beacon/AMZ5V2H6eeK0gGMjZhICzDhKXYhORQqXks5

bwdev01, 2016-11-13:
GA dialog layout revised to be approx 780w x 500h depending on # recommendations made.  Tested by OP.

closed 2016-11-13 by bwdev01