OpenPHDGuiding/phd2 #565: cam_QHY5.cpp file bug - maybe solved

astrovr, 2016-11-16:
Hi, I'm new on OpenPHDGuiding project so first of all congratulations to all contributors for this work!
I'm owner of a Magzero Mz5-m camera (QHY5 camera for italian market brand) that doesn't work well with PHD2 on Raspberry (ubuntu mate 16.04 OS): in particular calibration and guiding procedures fail (i use a 9x50 finder as guidescope and all calibration and guiding settings are correctly set).
After many test I've found the reason: a bug in cam_qhy5.cpp file.
The bug is here:

-----
`bool Camera_QHY5Class::ST4PulseGuideScope(int direction, int duration)
{
    int result = -1;
    int reg = 0;
    int32_t dur[2] = { -1, -1};

    duration /= 10;

    if (duration >= 255) duration = 254; // Max guide pulse is 2.54s -- 255 keeps it on always`

-----
with this settings calibration steps (and guiding corrections I suppose) are limited to 254 ms, a too short value for guiding expecially with short guidescopes (i.e. 9x50 finder that i use).
This bug is confirmed by testing the funcion "Manual guide" where even with long time pulses (1000 ms and longer) corresponds always a short noise of motor movements, compatible with a 254 ms correction pulse.
Also the calibration procedures confirm this: instead of about 12 steps, 40 or higher steps are necessary to perform (a bad) calibration.
I modify the cpp file as follows:

-----
` bool Camera_QHY5Class::ST4PulseGuideScope(int direction, int duration)
{
    int result = -1;
    int reg = 0;
    int32_t dur[2] = { -1, -1};

    // duration /= 10;

    if (duration >= 2500) duration = 2500; // Max guide pulse is 2.54s -- 255 keeps it on always `

----
and rebuild PHD2 according the procedure for Ubuntu.
Now calibration and guiding work. Calibration is done in about 12 steps and I don't lose the guiding star, even with a rough polar alignment.
In attachment the cam_qhy5.cpp file (you have to unzip the folder).
Sorry for my bad english...

Matteo

[cam_qhy5.zip](https://github.com/OpenPHDGuiding/phd2/files/594037/cam_qhy5.zip)

agalasso, 2016-11-16:
Matteo, great work, thanks for the fix!
The best way to get your change into the master branch would be for you to submit a new Pull Request.
Create your own fork containing your change, then submit the pull request.
Andy

astrovr, 2016-11-16:
Thanks Andy, i'll follow your instructions.
I forgot to explain how to control if the modification is properly installed.
1) Set the camera as QHY5 and mount On-camera.
2) open Manual Guide tab and set Guide Pulse duration first higher than 254 (i.e. 1500 ms), then with a smaller value (i.e. 250 ms). You'll be able to hear a clear different duration of motor noise in RA or DEC according the pulse direction tested.

Matteo

agalasso, 2016-11-16:
fixed by 6eb52e909c648ea867b3c1dba6f4e42b319ccc60

closed 2016-11-16 by agalasso