OpenPHDGuiding/phd2 #569: Directory structure cleanup

markus-k, 2016-12-05:
I mentioned this topic on IRC recently, but creating an issue for this is probably a good idea: Navigating the source code can be quite a pain with all the sources in the project root directory, mixed with a documentation and other bits and pieces. 

Apart from maybe removing some of the obsolete files (this is a version control system, they aren't lost if one ever needs them again), I would like to propose a directory layout something along those lines:
```
├── docs
│   └── help
│       └── index.html
├── lib
│   └── linux
│       └── i386
├── src
│   ├── cameras
│   │   ├── cam_INDI.cpp
│   │   └── cam_INDI.h
│   ├── gui
│   │   ├── indi_gui.cpp
│   │   └── indi_gui.h
│   ├── rotators
│   │   └── ...
│   ├── scopes
│   │   ├── scope_ASCOM.cpp
│   │   └── scope_ASCOM.h
│   ├── serialports
│   │   ├── ...
│   │   └── ...
│   ├── stepguiders
│   │   ├── ...
│   │   └── ...
│   ├── ...
│   ├── phd.cpp
│   └── phd.h
├── support
│   └── debian
└── README.md
```
This example doesn't cover everything, but gives a good idea. 

It might also be a good idea to keep build results like the PHD help zip out of the source tree and build in in a docs target.

I'd be happy to hear some feedback if you think this is a good idea, and why it might not be one.

raffienficiaud, 2016-12-05:
We are actively working on releasing the PEC control from branch `MPI-IS/learning_PEC_with_dithering`. Structuring better the repo is a good thing IMO (although I am not the one taking the decision), but it will create a lot of noise on our end.

markus-k, 2016-12-05:
Yes, other branches and prepared features should definitely be merged before restructuring the repository, would cause an unnecessary amount of pain otherwise.

agalasso, 2016-12-05:
agreed, we can revisit the idea of organizing the directory layout after the learning_PEC branch is merged. Hopefully we can make the changes incrementally to minimize the impact on the forks.

The subject of PHD help zip (and other generated files like .mo files) in the source tree is a separate issue.  The reason we have those in the source tree comes down to what requirements we are willing to impose on the Windows development environment.  The files are generated by perl and sh and other tools that are not available by default on Windows. Most Windows developers do not need to generate the help file so we are reluctant to ask them to install (e.g.) cygwin and/or perl just to build the help zip file.

d33psky, 2016-12-05:
I like the proposal (which markus already knew from our chat on irc :)
Agree on waiting for active merge-from-major-development branches.
Agree with the need to ease development on platforms.