OpenPHDGuiding/phd2 #572: "Identity" Guiding should not be called "None" in the UI

eklenske, 2016-12-15:
I know, this is a minor thing, but it confused me at first:

Under "Advanced Setup" (the Brain button) > "Algorithms", the first entry of the drop-down menu is called "None" in the UI:

    wxString xAlgorithms[] =
    {
        _("None"), _("Hysteresis"), _("Lowpass"), _("Lowpass2"), _("Resist Switch"),
    };

but in fact it enables `GUIDE_ALGORITHM_IDENTITY`, because the first element in the drop-down menu returns ID 0, which is the identity guiding:

    enum GUIDE_ALGORITHM
    {
        GUIDE_ALGORITHM_NONE=-1,
        GUIDE_ALGORITHM_IDENTITY,
        GUIDE_ALGORITHM_HYSTERESIS,
        GUIDE_ALGORITHM_LOWPASS,
        GUIDE_ALGORITHM_LOWPASS2,
        GUIDE_ALGORITHM_RESIST_SWITCH,
    };

So the question is: Should we support "None" as well (disable guiding entirely)? Then we should start `enum GUIDE_ALGORITHM` at 0 and add "Identity" to the drop-down menu. Or, should we keep the algorithms as is? Then we should change the string in the drop-down menu to "Identity" to prevent confusion.

agalasso, 2016-12-15:
We actually had it as "Identity" in the past and although that is technically a more precise term, it confused people; most people did not know what Identity meant.   I think we should leave it alone (having been through this already!).

Adding some information to a tool-tip explaining that None means identity would probably be an ok compromise.

Re: truly disabling guiding, we have that as a check-box item in the brain.  It makes sense to have this as a separate UI element since the likely use-case is to temporarily disable guiding to diagnose mount tracking/drift/etc.  When done the user re-enables guiding via the check-box, no messing around with the guide algo selection.

eklenske, 2016-12-15:
I see, thanks for the explanation, Andy!

One last thought: What do you think about calling it "Simple" or "Plain" instead of "None"?

agalasso, 2018-07-24:
obsoleted by 78ee9f0c3c52f43618d20dcf7d1549e5625ee93d (#713)

closed 2018-07-24 by agalasso