OpenPHDGuiding/phd2 #573: Linking fails with unresolvable R_X86_64_PC32 relocation

pauledd, 2016-12-21:
Hi
Trying to compile phd2 from source (latest git) on gentoo linux. No errors after "cmake .." but after 
"make -j4" I get this error:
```
[ 99%] Building CXX object CMakeFiles/phd2.dir/wxled.cpp.o
[100%] Linking CXX executable phd2
/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.3/../../../../x86_64-pc-linux-gnu/bin/ld: libcfitsio.a(eval_l.c.o)(.text+0xba2): unresolvable R_X86_64_PC32 relocation against symbol `ffleng'
/usr/lib/gcc/x86_64-pc-linux-gnu/4.9.3/../../../../x86_64-pc-linux-gnu/bin/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
CMakeFiles/phd2.dir/build.make:3039: recipe for target 'phd2' failed
make[2]: *** [phd2] Error 1
CMakeFiles/Makefile2:136: recipe for target 'CMakeFiles/phd2.dir/all' failed
make[1]: *** [CMakeFiles/phd2.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
```
Any ideas anyone?

agalasso, 2016-12-21:
not sure.  could you attach the output of a verbose build

`cd tmp; make clean all VERBOSE=1 2>&1 | tee logfile`

pauledd, 2016-12-22:
Sure:
http://pastebin.com/6BwrKv2G

Maybe it is expected to have cfitsio compiled with this "fpic" flag?

agalasso, 2016-12-23:
I do not think -fPIC is relevant here since libcfitsio.a is statically linked (not a dynamically loaded object).

Is it possible to try with gcc/g++ version 5 (or 6)?

pauledd, 2016-12-23:
I just tried gcc-5.4.0, same error plus another one complaining about indi "undefined reference to"
logfile:
http://pastebin.com/K4zUQQat

agalasso, 2016-12-24:
I'm running out of ideas. 

Here's another one: perhaps use your distro's cfitsio library rather than the one bundled with the phd2 sources?  In ubuntu-like distributions the package is libcfitsio-dev, not sure what the package is in gentoo.

pauledd, 2016-12-24:
it is already installed, it is called sci-libs/cfitsio version 3.360 with flags +fortran +tools -doc -examples -static-libs -threads

Does phd2 automaticaly recognize my system cfitsio or do I have to change something in the setup? I thought It always uses the directory that is in phd2...

agalasso, 2016-12-24:
phd2 uses the bundeled cfitsio, not the system version. To see if the system-installed version works you could grab the link line from the verbose log and substitute the system libcfitsio.a in place of the local one.  If your system file is /usr/lib64/libcfitsio.a  then I think you can replace "libcfitsio.a" with "/usr/lib64/libcfitsio.a"  like this:

`/usr/bin/c++    -std=c++11 -O3 -DNDEBUG   CMakeFiles/phd2.dir/mount.cpp.o CMakeFiles/phd2.dir/scope.cpp.o CMakeFiles/phd2.dir/scope_ascom.cpp.o CMakeFiles/phd2.dir/scope_eqmac.cpp.o CMakeFiles/phd2.dir/scope_equinox.cpp.o CMakeFiles/phd2.dir/scope_GC_USBST4.cpp.o CMakeFiles/phd2.dir/scope_gpint.cpp.o CMakeFiles/phd2.dir/scope_gpusb.cpp.o CMakeFiles/phd2.dir/scope_manual_pointing.cpp.o CMakeFiles/phd2.dir/scope_onboard_st4.cpp.o CMakeFiles/phd2.dir/scope_oncamera.cpp.o CMakeFiles/phd2.dir/scope_onstepguider.cpp.o CMakeFiles/phd2.dir/scope_voyager.cpp.o CMakeFiles/phd2.dir/scope_INDI.cpp.o CMakeFiles/phd2.dir/stepguider_sxao.cpp.o CMakeFiles/phd2.dir/stepguider_sxao_INDI.cpp.o CMakeFiles/phd2.dir/stepguider.cpp.o CMakeFiles/phd2.dir/cam_ascomlate.cpp.o CMakeFiles/phd2.dir/cam_Atik16.cpp.o CMakeFiles/phd2.dir/cam_firewire_IC.cpp.o CMakeFiles/phd2.dir/cam_firewire_OSX.cpp.o CMakeFiles/phd2.dir/cam_INDI.cpp.o CMakeFiles/phd2.dir/cam_INovaPLC.cpp.o CMakeFiles/phd2.dir/cam_KWIQGuider.cpp.o CMakeFiles/phd2.dir/cam_LEwebcam.cpp.o CMakeFiles/phd2.dir/cam_MeadeDSI.cpp.o CMakeFiles/phd2.dir/cam_NebSBIG.cpp.o CMakeFiles/phd2.dir/cam_opencv.cpp.o CMakeFiles/phd2.dir/cam_openssag.cpp.o CMakeFiles/phd2.dir/cam_OSPL130.cpp.o CMakeFiles/phd2.dir/cam_QGuide.cpp.o CMakeFiles/phd2.dir/cam_QHY5II.cpp.o CMakeFiles/phd2.dir/cam_QHY5LII.cpp.o CMakeFiles/phd2.dir/cam_SAC42.cpp.o CMakeFiles/phd2.dir/cam_SACGuide.cpp.o CMakeFiles/phd2.dir/cam_SBIG.cpp.o CMakeFiles/phd2.dir/cam_sbigrotator.cpp.o CMakeFiles/phd2.dir/cam_simulator.cpp.o CMakeFiles/phd2.dir/cam_SSAG.cpp.o CMakeFiles/phd2.dir/cam_SSPIAG.cpp.o CMakeFiles/phd2.dir/cam_Starfish.cpp.o CMakeFiles/phd2.dir/cam_Altair.cpp.o CMakeFiles/phd2.dir/cam_StarShootDSCI.cpp.o CMakeFiles/phd2.dir/cam_SXV.cpp.o CMakeFiles/phd2.dir/cam_VFW.cpp.o CMakeFiles/phd2.dir/cam_wdm.cpp.o CMakeFiles/phd2.dir/cam_ZWO.cpp.o CMakeFiles/phd2.dir/camera.cpp.o CMakeFiles/phd2.dir/cam_qhy5.cpp.o CMakeFiles/phd2.dir/cameras/SXMacLib.c.o CMakeFiles/phd2.dir/backlash_comp.cpp.o CMakeFiles/phd2.dir/guide_algorithm_hysteresis.cpp.o CMakeFiles/phd2.dir/guide_algorithm_identity.cpp.o CMakeFiles/phd2.dir/guide_algorithm_lowpass.cpp.o CMakeFiles/phd2.dir/guide_algorithm_lowpass2.cpp.o CMakeFiles/phd2.dir/guide_algorithm_resistswitch.cpp.o CMakeFiles/phd2.dir/guide_algorithm.cpp.o CMakeFiles/phd2.dir/guider_onestar.cpp.o CMakeFiles/phd2.dir/guider.cpp.o CMakeFiles/phd2.dir/about_dialog.cpp.o CMakeFiles/phd2.dir/advanced_dialog.cpp.o CMakeFiles/phd2.dir/aui_controls.cpp.o CMakeFiles/phd2.dir/calreview_dialog.cpp.o CMakeFiles/phd2.dir/calstep_dialog.cpp.o CMakeFiles/phd2.dir/camcal_import_dialog.cpp.o CMakeFiles/phd2.dir/comet_tool.cpp.o CMakeFiles/phd2.dir/configdialog.cpp.o CMakeFiles/phd2.dir/confirm_dialog.cpp.o CMakeFiles/phd2.dir/darks_dialog.cpp.o CMakeFiles/phd2.dir/debuglog.cpp.o CMakeFiles/phd2.dir/drift_tool.cpp.o CMakeFiles/phd2.dir/eegg.cpp.o CMakeFiles/phd2.dir/event_server.cpp.o CMakeFiles/phd2.dir/fitsiowrap.cpp.o CMakeFiles/phd2.dir/gear_dialog.cpp.o CMakeFiles/phd2.dir/graph-stepguider.cpp.o CMakeFiles/phd2.dir/graph.cpp.o CMakeFiles/phd2.dir/guiding_assistant.cpp.o CMakeFiles/phd2.dir/guidinglog.cpp.o CMakeFiles/phd2.dir/image_math.cpp.o CMakeFiles/phd2.dir/json_parser.cpp.o CMakeFiles/phd2.dir/logger.cpp.o CMakeFiles/phd2.dir/manualcal_dialog.cpp.o CMakeFiles/phd2.dir/messagebox_proxy.cpp.o CMakeFiles/phd2.dir/myframe.cpp.o CMakeFiles/phd2.dir/myframe_events.cpp.o CMakeFiles/phd2.dir/nudge_lock.cpp.o CMakeFiles/phd2.dir/onboard_st4.cpp.o CMakeFiles/phd2.dir/optionsbutton.cpp.o CMakeFiles/phd2.dir/phd.cpp.o CMakeFiles/phd2.dir/phdconfig.cpp.o CMakeFiles/phd2.dir/phdcontrol.cpp.o CMakeFiles/phd2.dir/profile_wizard.cpp.o CMakeFiles/phd2.dir/Refine_DefMap.cpp.o CMakeFiles/phd2.dir/rotator.cpp.o CMakeFiles/phd2.dir/rotator_ascom.cpp.o CMakeFiles/phd2.dir/runinbg.cpp.o CMakeFiles/phd2.dir/serialport.cpp.o CMakeFiles/phd2.dir/serialport_loopback.cpp.o CMakeFiles/phd2.dir/serialport_mac.cpp.o CMakeFiles/phd2.dir/serialport_win32.cpp.o CMakeFiles/phd2.dir/serialport_posix.cpp.o CMakeFiles/phd2.dir/socket_server.cpp.o CMakeFiles/phd2.dir/starcross_test.cpp.o CMakeFiles/phd2.dir/statswindow.cpp.o CMakeFiles/phd2.dir/star.cpp.o CMakeFiles/phd2.dir/star_profile.cpp.o CMakeFiles/phd2.dir/target.cpp.o CMakeFiles/phd2.dir/testguide.cpp.o CMakeFiles/phd2.dir/usImage.cpp.o CMakeFiles/phd2.dir/worker_thread.cpp.o CMakeFiles/phd2.dir/config_INDI.cpp.o CMakeFiles/phd2.dir/indi_gui.cpp.o CMakeFiles/phd2.dir/wxled.cpp.o  -o phd2 -rdynamic -lX11 /usr/lib64/libcfitsio.a -lusb-1.0 -L/usr/lib64 -pthread -lwx_gtk2u_aui-3.0 -lwx_gtk2u_core-3.0 -lwx_baseu-3.0 -lwx_gtk2u_adv-3.0 -lwx_gtk2u_html-3.0 -lwx_baseu_net-3.0 ../cameras/zwolibs/x64/libASICamera2.a -lm -Wl,-Bstatic -lindiclient -Wl,-Bdynamic -lindi -lz -lnova -lm`

pauledd, 2016-12-25:
Unfortunately your suggestion did not work. BUT it compiled now :)  I cant remember what I did exactly but I did: update libnova to 0.15.0, rebuilt my system cfitsio with enabled flags "bzip static-libs threads", deleted whole phd2 dir and cloned a fresh one from git.
Nevertheless thanks a lot for your effort to help! 
greetings and a happy new year
Paul

closed 2016-12-25 by pauledd