OpenPHDGuiding/phd2 #575: add server api method to read guide camera CCD temperature

agalasso, 2017-01-08:
request to record guide camera temperature in guide log, conversation [here](https://groups.google.com/d/topic/open-phd-guiding/3jU9QLXNstM/discussion)

in lieu of modifying the guide log format to accommodate this, we can add a method to the server api so the application can access the ccd temperature info via the server API

deix, 2017-10-04:
Hi there,

this would be a very nice and welcomed enhancement. Looking at the "ASICamera2 Software Development Kit.pdf" from "SDK_V1.13.1.4" on page 5 we can find the necessary entry to get sensor temperature: "ASI_TEMPERATURE,// sensor temperature，10 times the actual temperature". Hope this can be implemented soon in PHD2.

Best regards
Michal

agalasso, 2017-10-05:
fixed by e8bad0a5d25508476b54a302a1e8711037472a7b

closed 2017-10-05 by agalasso