OpenPHDGuiding/phd2 #578: Overriding users chioce of install directory

Sector14, 2017-01-26:
The current CMakeLists.txt overrides the users choice of install directory to /usr Is there any chance this could be amended to just default to /usr only when the user has not selected a different directory? For example

``
if(UNIX AND NOT APPLE)
  set(CMAKE_INSTALL_PREFIX /usr CACHE PATH "PHD2 install prefix")
endif()
``

Would allow the install prefix to be overridden by the user as this is set above the project() line. An alternative is to move this all below the project and use

``
if (CMAKE_INSTALL_PREFIX_INITIALIZED_TO_DEFAULT)
    set (CMAKE_INSTALL_PREFIX /usr CACHE PATH "PHD2 install path" FORCE )
endif()
``

It would save those who wish to keep non distro installs in /usr/local from editing the CMakeLists.txt or as in my case, from seeing the CMAKE_INSTALL_PREFIX reported as /usr/local then getting a bit of a surprise when it installs into /usr

Although I've only tested the first option, the latter option is the one [recommended by the cmake devs ](http://public.kitware.com/pipermail/cmake/2010-December/041135.html)

d33psky, 2017-03-03:
Yes having a configurable install prefix makes total sense.
The CMAKE_INSTALL_PREFIX_INITIALIZED_TO_DEFAULT looks awkward, even though it was recommended in 2010. Is that still 'the way to go' ?

d33psky, 2017-03-03:
nvm, read up on this thing and it looks good. Testing now.

d33psky, 2017-03-03:
Tested and applied.
https://github.com/OpenPHDGuiding/phd2/commit/47f54eacc61115711a9bbb567fe4c5091ef1b67c
Thanks for the enhancement !

closed 2017-03-03 by d33psky