OpenPHDGuiding/phd2 #580: missing camera controls on camera tab of brain before camera connected causes confusion

agalasso, 2017-02-05:
Some camera properties like binning and subframes have controls that show up in the camera tab of the advanced dialog only after the camera is connected. This sometimes causes confusion when users go looking for the controls when the camera is not connected.

We should address this, perhaps by having the controls always present but disabled when the camera is not connected. And/or display some text in the window indicating that the camera controls will be available after the camera is connected.

d33psky, 2017-03-03:
I propose to do the latter "display some text in the window indicating that the camera controls will be available after the camera is connected". (And maybe add a green light stating the camera is connected when it is.)
This also lines up well with INDI device property discovery.

bwdev01, 2017-03-03:
> brain before camera connected causes confusion (#580)

There's been some preliminary thinking done on this for Windows.  There's
already a camera icon in the status bar so I don't think we need more
lights.  The current behavior does seem to cause some confusion so we'll
need to make some kind of change.
 
  _____  

it on GitHub, or mute
5CA1kn4ks5riAmfgaJpZM4L3pxv>  the thread.
<https://github.com/notifications/beacon/AMZ5Vxf0zuqSZBZl5GQ7lHS5yfK-3nxzks5
riAmfgaJpZM4L3pxv.gif>

bwdev01, 2017-03-12:
Advisory text block is now shown when the camera is not connected.  Post 2.6.3 completion.

closed 2017-03-12 by bwdev01