OpenPHDGuiding/phd2 #590: guiding assistant table layout incorrect for translated version(s)

agalasso, 2017-04-09:
labels need to wrap, or the window needs to be re-sizable, or something

[image]

bwdev01, 2017-04-09:
Do we know the translation was done sensibly, taking the particular use into account?  Does 'RA max drift rate' really require over 50 character positions?  This dialog is heavily constrained by the need to support small screens.  If this can't be resolved with leaner translations, we may need to use a fully scrollable window.

ristirahva, 2017-04-11:
Hi,

I can translate  'RA max drift rate' like 'Макс. скорость дрейфа по RA' but
it looks a bit unreadable.
I think it would be better to make columns resizeable.

agalasso, 2017-04-11:
resizing the first column to fit the text gives this:

[image]

If that is acceptable, I'll commit the change.  It would probably be a good idea to edit the translation of row 5 to make it shorter if possible.

bwdev01, 2017-04-11:
But now you've got a dialog box that appears to be something like 950px wide
and we're going to get complaints on that front.  Why don't we just set the
option to EnableDragColSize on these grids. Then the user can resize them if
needed.  If a column is widened, a horizontal scroll bar appears on the
grid, so the info in all the columns can always be read and the overall
dimensions of the dialog aren't affected. 
 
Bruce
 
  _____  

for translated version(s) (#590)
 

 
<https://cloud.githubusercontent.com/assets/6864470/24929477/0b4f3a58-1ed4-1
1e7-9ce0-fbccc306a70d.png> image
it on GitHub, or mute
ZOcG7-1ks5ru-MjgaJpZM4M36nY>  the thread.
<https://github.com/notifications/beacon/AMZ5V-hnz4mVQwnLKOakpxdqket1OC-Bks5
ru-MjgaJpZM4M36nY.gif>

agalasso, 2017-04-11:
> Why don't we just set the
option to EnableDragColSize on these grids. Then the user can resize them if
needed.

I believe column resizing is enabled already .. it works for me with the latest code on Windows and Linux.

agalasso, 2017-04-11:
Here is how it looks with text wrapping. Acceptible?

[image]

ristirahva, 2017-04-11:
Looks nice. Thanks!

agalasso, 2017-04-25:
fixed by 0c7498ac9de727e6692faa409fab20ea8adcb691

closed 2017-04-25 by agalasso