OpenPHDGuiding/phd2 #591: Build broken on debian/jessie64; fails to link with latest libqhy

freerobby, 2017-04-10:
Last night I tried to build the latest version of PHD2 on debian linux, and it failed during linking. I did a git bisect and found the breaking change was when the new QHY libraries (#589) were merged in.

```
[snip]
[ 96%] Building CXX object CMakeFiles/phd2.dir/star.cpp.o
[ 96%] Building CXX object CMakeFiles/phd2.dir/star_profile.cpp.o
[ 97%] Building CXX object CMakeFiles/phd2.dir/target.cpp.o
[ 97%] Building CXX object CMakeFiles/phd2.dir/testguide.cpp.o
[ 98%] Building CXX object CMakeFiles/phd2.dir/usImage.cpp.o
[ 98%] Building CXX object CMakeFiles/phd2.dir/worker_thread.cpp.o
[ 99%] Building CXX object CMakeFiles/phd2.dir/config_INDI.cpp.o
[ 99%] Building CXX object CMakeFiles/phd2.dir/indi_gui.cpp.o
[100%] Building CXX object CMakeFiles/phd2.dir/wxled.cpp.o
Linking CXX executable phd2
../cameras/qhyccdlibs/linux/x86_64/libqhy.a(log4z.o): In function `zsummer::log4z::fixPath(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)':
log4z.cpp:(.text+0x164): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::empty() const'
log4z.cpp:(.text+0x178): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::begin()'
log4z.cpp:(.text+0x188): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::end()'
log4z.cpp:(.text+0x1e4): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::length() const'
log4z.cpp:(.text+0x1f7): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::at(unsigned long)'
log4z.cpp:(.text+0x214): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::append(char const*)'
../cameras/qhyccdlibs/linux/x86_64/libqhy.a(log4z.o): In function `zsummer::log4z::trimLogConfig(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
log4z.cpp:(.text+0x259): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::empty() const'
log4z.cpp:(.text+0x272): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator+=(char const*)'
log4z.cpp:(.text+0x27e): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::length() const'
log4z.cpp:(.text+0x29b): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::length() const'
log4z.cpp:(.text+0x2b2): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator[](unsigned long)'
log4z.cpp:(.text+0x2ca): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator[](unsigned long)'
log4z.cpp:(.text+0x2e2): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator[](unsigned long)'
log4z.cpp:(.text+0x329): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::length() const'
log4z.cpp:(.text+0x348): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator[](unsigned long)'
log4z.cpp:(.text+0x360): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator[](unsigned long)'
log4z.cpp:(.text+0x3c3): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::substr(unsigned long, unsigned long) const'
log4z.cpp:(.text+0x3d6): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator=(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
log4z.cpp:(.text+0x3e2): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'
log4z.cpp:(.text+0x3f0): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::clear()'
log4z.cpp:(.text+0x401): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'
../cameras/qhyccdlibs/linux/x86_64/libqhy.a(log4z.o): In function `zsummer::log4z::splitPairString(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
log4z.cpp:(.text+0x461): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::c_str() const'
log4z.cpp:(.text+0x478): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::find(char const*, unsigned long) const'
log4z.cpp:(.text+0x496): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
log4z.cpp:(.text+0x4da): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'
log4z.cpp:(.text+0x4ee): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::length() const'
log4z.cpp:(.text+0x50f): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::substr(unsigned long, unsigned long) const'
log4z.cpp:(.text+0x52b): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::substr(unsigned long, unsigned long) const'
log4z.cpp:(.text+0x54f): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'
log4z.cpp:(.text+0x55b): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'
log4z.cpp:(.text+0x57d): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'
log4z.cpp:(.text+0x597): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'
log4z.cpp:(.text+0x5a8): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'
../cameras/qhyccdlibs/linux/x86_64/libqhy.a(log4z.o): In function `zsummer::log4z::parseConfigLine(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, zsummer::log4z::LoggerInfo, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, zsummer::log4z::LoggerInfo> > >&)':
log4z.cpp:(.text+0x637): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(char const*, std::allocator<char> const&)'
log4z.cpp:(.text+0x666): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'
log4z.cpp:(.text+0x684): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::empty() const'
log4z.cpp:(.text+0x6a1): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string()'
log4z.cpp:(.text+0x6c9): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'
log4z.cpp:(.text+0x6d8): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string()'
log4z.cpp:(.text+0x704): undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()'
[... cut ...]
Last working build is against this tree: https://github.com/OpenPHDGuiding/phd2/tree/ce42e6dc7d833c24c46be168f87e6400e12aff99

Some environmental info is below.

```
root@66ed246f39da:/phd2/tmp# uname -a
Linux 66ed246f39da 3.16.0-4-amd64 #1 SMP Debian 3.16.36-1+deb8u1 (2016-09-03) x86_64 GNU/Linux

root@66ed246f39da:/phd2/tmp# gcc --version
gcc (Debian 4.9.2-10) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

agalasso, 2017-04-10:
The new QHY SDK has been temporarily disabled for Linux until we get an updated SDK library from QHY that resolves the link dependency. QHY has been informed about the issue.

agalasso, 2017-04-10:
Note: to re-enable the SDK in Linux: revert the change thats disabled the SDK: eb806c174e04e310e455f7e6b19a2d712f297f4c

agalasso, 2017-04-19:
@freerobby - we got an updated SDK from QHY that should fix the problem. Could you try pulling the latest code and re-building to confirm it is fixed?  no need to edit cameras.h, just pull and build.

freerobby, 2017-04-20:
👍 Fixed. Thanks!

closed 2017-04-20 by freerobby