OpenPHDGuiding/phd2 #595: Master branch fails to compile

lock042, 2017-04-22:
after a 
`git pull`
from today, phd2 fails to compile. Here the log:
`CMakeFiles/phd2.dir/cam_qhy.cpp.o: In function `Camera_QHY::~Camera_QHY()':
cam_qhy.cpp:(.text+0x59): undefined reference to `ReleaseQHYCCDResource'
CMakeFiles/phd2.dir/cam_qhy.cpp.o: In function `Camera_QHY::Disconnect()':
cam_qhy.cpp:(.text+0x7c): undefined reference to `CancelQHYCCDExposingAndReadout'
cam_qhy.cpp:(.text+0x88): undefined reference to `CloseQHYCCD'
CMakeFiles/phd2.dir/cam_qhy.cpp.o: In function `Camera_QHY::ST4PulseGuideScope(int, int)':
cam_qhy.cpp:(.text+0xe6): undefined reference to `ControlQHYCCDGuide'
CMakeFiles/phd2.dir/cam_qhy.cpp.o: In function `Camera_QHY::~Camera_QHY()':
cam_qhy.cpp:(.text+0x159): undefined reference to `ReleaseQHYCCDResource'
CMakeFiles/phd2.dir/cam_qhy.cpp.o: In function `Camera_QHY::Capture(int, usImage&, int, wxRect const&)':
cam_qhy.cpp:(.text+0x5b4): undefined reference to `SetQHYCCDBinMode'
cam_qhy.cpp:(.text+0x5dd): undefined reference to `CancelQHYCCDExposingAndReadout'
cam_qhy.cpp:(.text+0x7c7): undefined reference to `SetQHYCCDResolution'
cam_qhy.cpp:(.text+0x82b): undefined reference to `SetQHYCCDParam'
cam_qhy.cpp:(.text+0xbe9): undefined reference to `SetQHYCCDParam'
cam_qhy.cpp:(.text+0xc0c): undefined reference to `ExpQHYCCDSingleFrame'
cam_qhy.cpp:(.text+0xe38): undefined reference to `GetQHYCCDSingleFrame'
CMakeFiles/phd2.dir/cam_qhy.cpp.o: In function `Camera_QHY::Connect(wxString const&)':
cam_qhy.cpp:(.text+0x39c9): undefined reference to `ScanQHYCCD'
cam_qhy.cpp:(.text+0x3a51): undefined reference to `IsQHYCCDControlAvailable'
cam_qhy.cpp:(.text+0x4005): undefined reference to `GetQHYCCDId'
cam_qhy.cpp:(.text+0x400d): undefined reference to `OpenQHYCCD'
cam_qhy.cpp:(.text+0x43a1): undefined reference to `InitQHYCCDResource'
cam_qhy.cpp:(.text+0x4702): undefined reference to `OpenQHYCCD'
cam_qhy.cpp:(.text+0x484b): undefined reference to `SetQHYCCDStreamMode'
cam_qhy.cpp:(.text+0x4866): undefined reference to `CloseQHYCCD'
cam_qhy.cpp:(.text+0x4b0e): undefined reference to `InitQHYCCD'
cam_qhy.cpp:(.text+0x4b22): undefined reference to `CloseQHYCCD'
cam_qhy.cpp:(.text+0x5382): undefined reference to `GetQHYCCDParamMinMaxStep'
cam_qhy.cpp:(.text+0x53d0): undefined reference to `GetQHYCCDChipInfo'
cam_qhy.cpp:(.text+0x53e8): undefined reference to `CloseQHYCCD'
cam_qhy.cpp:(.text+0x5618): undefined reference to `CloseQHYCCD'
cam_qhy.cpp:(.text+0x5e0b): undefined reference to `IsQHYCCDControlAvailable'
cam_qhy.cpp:(.text+0x5f7d): undefined reference to `IsQHYCCDControlAvailable'
cam_qhy.cpp:(.text+0x61dd): undefined reference to `SetQHYCCDBinMode'
cam_qhy.cpp:(.text+0x624e): undefined reference to `GetQHYCCDMemLength'
cam_qhy.cpp:(.text+0x63d3): undefined reference to `SetQHYCCDResolution'
cam_qhy.cpp:(.text+0x63e7): undefined reference to `CloseQHYCCD'
cam_qhy.cpp:(.text+0x6502): undefined reference to `CloseQHYCCD'
collect2: error: ld returned 1 exit status
CMakeFiles/phd2.dir/build.make:3070: recipe for target 'phd2' failed
make[2]: *** [phd2] Error 1
CMakeFiles/Makefile2:177: recipe for target 'CMakeFiles/phd2.dir/all' failed
make[1]: *** [CMakeFiles/phd2.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
`

agalasso, 2017-04-22:
help us out a little here... is this a Linux build? What distro and version and what architecture?
uname -a ?

lock042, 2017-04-22:
Oh, I'm sorry.
Yes I'm running on linux. Debian testing.
`Linux debian 4.9.0-2-amd64 #1 SMP Debian 4.9.18-1 (2017-03-30) x86_64 GNU/Linux`

agalasso, 2017-04-22:
thx. What version of Debian is that?
lsb_release -a

lock042, 2017-04-23:
`lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.0 (stretch)
Release:	9.0
Codename:	stretch`

agalasso, 2017-04-23:
thanks.  I have contacted QHY requesting an updated libqhy.a that is compatible with Debian stretch.
In the meantime, if you are not using a QHY camera you can apply this patch and rebuild:

```
--- a/cameras.h
+++ b/cameras.h
@@ -104,7 +104,7 @@
 # define SIMULATOR
 # define CAM_QHY5
 #ifndef __arm__     // until we get an updated SDK from QHY uncluding arm version(s) of libqhy.a
-# define QHY_CAMERA
+//# define QHY_CAMERA
 #endif
 # define INDI_CAMERA
 # define ZWO_ASI

```

lock042, 2017-04-23:
Thank you so much.
I will do that

agalasso, 2017-04-23:
detailed log attached to #596

agalasso, 2017-04-26:
fixed by 467b7ba907b58c5815ac2df4fb165aa021a1dca7

closed 2017-04-26 by agalasso