OpenPHDGuiding/phd2 #597: no icon visible on Debian Stretch

lock042, 2017-04-23:
phd2 master has no icon available
I have this warning:
http://hpics.li/d89491e
and this result:
http://hpics.li/002d1fa

On the console I have:

(phd2:3656): GdkPixbuf-CRITICAL **: gdk_pixbuf_get_height: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

** (phd2:3656): WARNING **: Invalid borders specified for theme pixmap:
        /usr/share/themes/Adwaita/gtk-2.0/assets/line.png,
borders don't fit within the image

(phd2:3656): GdkPixbuf-CRITICAL **: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(phd2:3656): GdkPixbuf-CRITICAL **: gdk_pixbuf_get_pixels_with_length: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(phd2:3656): GdkPixbuf-CRITICAL **: gdk_pixbuf_get_rowstride: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

agalasso, 2017-04-23:
I cannot reproduce the PNG load error on Debian 9.0 (stretch). Are you really using stretch, or perhaps are you using Debian testing?

The GdkPixbuf warnings are a red herring, we have always had those and they are harmless.

lock042, 2017-04-23:
I am on testing, but testing is on freeze phase.
So yet testing=stretch.

agalasso, 2017-04-23:
well there must be some difference between testing and stretch since it works fine on plain stretch.

maybe there is a wxWidgets version difference?  that will show in the debug log... could you attach your PHD2 debug log ($HOME/PHD2/PHD2_DebugLog_<date-time>.txt)

lock042, 2017-04-23:
Here the log:
https://free-astro.org/download/PHD2_DebugLog_2017-04-23_092034.txt

agalasso, 2017-04-23:
thanks for the log.  that's the same version of wx that I have.

FWIW, the version that I installed and is working is [Debian Installer Stretch RC 3 release](https://www.debian.org/devel/debian-installer/)

Is that what you are using?

lock042, 2017-04-23:
The only thing I can say is that I'm using testing as you can see on my source file

deb http://ftp2.fr.debian.org/debian/ testing main contrib non-free
deb-src http://ftp2.fr.debian.org/debian/ testing main contrib non-free

deb http://security.debian.org/ testing/updates main
deb-src http://security.debian.org/ testing/updates main

# testing-updates, previously known as 'volatile'
deb http://ftp2.fr.debian.org/debian/ testing-updates main
deb-src http://ftp2.fr.debian.org/debian/ testing-updates main

agalasso, 2017-04-23:
mine is:

deb http://ftp.us.debian.org/debian/ stretch main
deb-src http://ftp.us.debian.org/debian/ stretch main

deb http://security.debian.org/debian-security stretch/updates main
deb-src http://security.debian.org/debian-security stretch/updates main

lock042, 2017-04-23:
Yes. And repositories should be the same ...

agalasso, 2017-04-23:
are you up-to-date on everything? (apt-get update; apt-get upgrade)
... just trying to get us on the same page so I can reproduce the issue

agalasso, 2017-04-23:
regarding the PNG load error image you posted earlier, there is a "Details" option in the error dialog... was there any information there that you could post here?

lock042, 2017-04-23:
Yes. Everything is up to date on my system.
I do it everyday.
Sorry for not helping you much better

lock042, 2017-04-23:
I could try to execute phd2 with gdb

agalasso, 2017-04-23:
sure, that might help.  You may need to rebuild it with debug symbols, not sure if they are enabled by default:
```
mkdir debug
cd debug
cmake -DCMAKE_BUILD_TYPE=Debug ..
make
gdb ./phd2

```
also, when you get a chance could you provide the information from the Details of the error window as requested above

lock042, 2017-04-24:
Of course.

Here the log:

11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:43:36: Application built with libpng-1.6.28 but running with 1.5.12
11:43:36: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
[... 1 lines cut ...]
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
[... cut ...]
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.
11:44:18: Application built with libpng-1.6.28 but running with 1.5.12
11:44:18: Couldn't load a PNG image - file is corrupted or not enough memory.

pchev, 2017-04-24:
libpng version on Stretch is 1.6.28, your problem is probably because of this old 1.5.12 somewhere in your ldconfig search path. 

You can try to find the old library with :  locate libpng.so
then to check the version run the following for each: 
readelf -d /usr/lib/x86_64-linux-gnu/libpng.so | grep SONAME

lock042, 2017-04-24:
`
LANG=C readelf -d /usr/lib/x86_64-linux-gnu/libpng.so | grep SONAME
 0x000000000000000e (SONAME)             Library soname: [libpng16.so.16]
`

pchev, 2017-04-24:
Yes, this one is the right one but you have somewhere a old library with version 1.5.12 as show in your log. 
Try :  locate libpng.so
and repeat the readelf command for each result path.

A classic error is you have a old libpng.so.*  in /usr/local/lib/ because you complied this library some time ago.

lock042, 2017-04-24:
OK. The only other libpng I have is not use by something else except PixInsight.

> lock@debian:~/Documents/phd2/debug$ locate libpng.so
> /home/lock/Documents/PI/PI-linux-x86_64-01.08.04.1198-20160221-c/PixInsight/bin/lib/libpng.so
> /opt/PixInsight/bin/lib/libpng.so
> /usr/lib/x86_64-linux-gnu/libpng.so
> lock@debian:~/Documents/phd2/debug$ LANG=C readelf -d /opt/PixInsight/bin/lib/libpng.so | grep SONAME
>  0x000000000000000e (SONAME)             Library soname: [libpng15.so.15]

```
lock@debian:~/Documents/phd2/debug$ ll /opt/PixInsight/bin/lib/libpng.so
lrwxrwxrwx 1 root root 11 oct.   9  2016 /opt/PixInsight/bin/lib/libpng.so -> libpng15.so
lock@debian:~/Documents/phd2/debug$ ll /usr/lib/x86_64-linux-gnu/libpng.so
lrwxrwxrwx 1 root root 11 janv.  9 19:50 /usr/lib/x86_64-linux-gnu/libpng.so -> libpng16.so
lock@debian:~/Documents/phd2/debug$ ll /usr/lib/x86_64-linux-gnu/libpng.so
lrwxrwxrwx 1 root root 11 janv.  9 19:50 /usr/lib/x86_64-linux-gnu/libpng.so -> libpng16.so
```

agalasso, 2017-04-24:
`ldd ./phd2 | grep libpng` should help you identify whether phd2 is picking up the PixInsight version. I would only think that would happen if you had LD_LIBRARY_PATH set to include /opt/PixInsight/bin/lib -- is that the case? (`echo $LD_LIBRARY_PATH`)

agalasso, 2017-04-24:
perhaps also try `ldconfig -v` and see what that shows for libpng ?

lock042, 2017-04-24:
```
LANG=C ldd ./phd2 | grep libpng
libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007ff12151e000)
```
and $LD_LIBRARY_PATH is empty.
and ldconfig -v | grep libpng

```
ldconfig: Ne peut évaluer par stat /lib/i686-linux-gnu: Aucun fichier ou dossier de ce type
ldconfig: Ne peut évaluer par stat /usr/lib/i686-linux-gnu: Aucun fichier ou dossier de ce type
ldconfig: Chemin « /lib/x86_64-linux-gnu » donné plus d'une fois
ldconfig: Chemin « /usr/lib/x86_64-linux-gnu » donné plus d'une fois
ldconfig: /lib/i386-linux-gnu/ld-2.24.so is the dynamic linker, ignoring

	libpng16.so.16 -> libpng16.so.16.28.0
ldconfig: /lib/x86_64-linux-gnu/ld-2.24.so is the dynamic linker, ignoring

	libpng16.so.16 -> libpng16.so.16.28.0
ldconfig: /lib32/ld-2.24.so is the dynamic linker, ignoring

ldconfig: /libx32/ld-2.24.so is the dynamic linker, ignoring

```

lock042, 2017-04-25:
So phd takes the right library ....

pchev, 2017-04-25:
It should , but you log show "... running with 1.5.12" so the is something wrong with your libpng setting.

Can you try to reinstall the libpng packages:
apt-get install --reinstall libpng-dev libpng16-16

Then rebuild phd2 clean, including running cmake.

lock042, 2017-04-25:
As I said, i already did it.
But noting works :(.

I have no icons in phd2 :(.

lock042, 2017-04-25:
In my log it is libpng 1.5.12. But PixInsight only use 1.5.13. I have no 1.5.12 on my machine.
```
locate libpng15.so.15*
/home/lock/Documents/PI/PI-linux-x86_64-01.08.04.1198-20160221-c/PixInsight/bin/lib/libpng15.so.15
/home/lock/Documents/PI/PI-linux-x86_64-01.08.04.1198-20160221-c/PixInsight/bin/lib/libpng15.so.15.13.0
/opt/PixInsight/bin/lib/libpng15.so.15
/opt/PixInsight/bin/lib/libpng15.so.15.13.0
```

agalasso, 2017-04-25:
@lock042 could you try this:

```
cd <phd2_source_dir>
rm -rf tmp
mkdir tmp
cd tmp
( cmake ..  && make VERBOSE=1 ) 2>&1 | tee build-log
```

then attach the file build-log

lock042, 2017-04-25:
I also tried to remove the whole directory and I re-cloned the git repository with no success.

[build-log.txt](https://github.com/OpenPHDGuiding/phd2/files/956403/build-log.txt)

agalasso, 2017-04-25:
ok, thanks for that log. I'm still stumped why it works for me but not for you.
How about this info:
`apt list --installed | grep wx`

lock042, 2017-04-25:
```
apt list --installed | grep wx

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libwxbase3.0-0v5/testing,now 3.0.2+dfsg-4 amd64  [installé, automatique]
libwxbase3.0-dev/testing,now 3.0.2+dfsg-4 amd64  [installé, automatique]
libwxgtk3.0-0v5/testing,now 3.0.2+dfsg-4 amd64  [installé, automatique]
libwxgtk3.0-dev/testing,now 3.0.2+dfsg-4 amd64  [installé]
plplot12-driver-wxwidgets/testing,now 5.10.0+dfsg2-0.4+b2 amd64  [installé, automatique]
python-wxgtk3.0/testing,now 3.0.2.0+dfsg-4 amd64  [installé, automatique]
python-wxtools/testing,testing,now 3.0.2.0+dfsg-4 all  [installé, automatique]
python-wxversion/testing,testing,now 3.0.2.0+dfsg-4 all  [installé, automatique]
wx-common/testing,now 3.0.2+dfsg-4 amd64  [installé]
wx3.0-headers/testing,now 3.0.2+dfsg-4 amd64  [installé, automatique]
wx3.0-i18n/testing,testing,now 3.0.2+dfsg-4 all  [installé]
wxhexeditor/testing,now 0.23+repack-2+b2 amd64  [installé]

```

agalasso, 2017-04-26:
@lock042  are you proficient with gdb? I think the next thing for us to do is to step through wxPNGHandler::LoadFile in gdb and see where the failure is coming from. 
```
cd $PHD_SRC/tmp
sudo apt-get install libwxbase3.0-0v5-dbg
apt-get source libwxbase3.0-dev
gdb ./phd2
(gdb) dir wxwidgets3.0-3.0.2+dfsg/src
(gdb) tb main
(gdb) run
(gdb) b wxPNGHandler::LoadFile
(gdb) c
<step through code until you get to "goto error", which part failed?>
```
We don't have a lot of options other than to debug it right on your system since we cannot reproduce it elsewhere.
If you want, I could take a look at it in a screen session via ssh - if you want to try that, drop me an email andy.galasso@gmail.com

lock042, 2017-04-26:
Oh, I see you did reproduced the bug in #592 ??.
Good.

If you want I can give you a session by teamviewer ?

agalasso, 2017-04-26:
it was not reproduced by me, that was a separate report (in another environment that I do not have access to (RPi)).
sure, I can do teamviewer, please email me at the email address above

commit ef628f0a (2017-04-26): linux: prevent cmake from picking up a libqhy.so installed on the system - fixes #597
closed 2017-04-26 by agalasso