OpenPHDGuiding/phd2 #599: PPEC: phd2 freezes upon calibration completion with Simulator

domokos, 2017-04-25:
phd2 2.6.3
built on Opensuse Leap 42.2
Works fine if built witout PPEC option.

If built with cmake flag -DGUIDING_GAUSSIAN_PROCESS=true then after completing calibration it freezes.

doma@domalaptop:~/git/phd2> ldd /usr/local/bin/phd2
        linux-vdso.so.1 (0x00007fffb1df7000)
        libX11.so.6 => /usr/lib64/libX11.so.6 (0x00007fcf28f95000)
        libwx_gtk2u_aui-suse.so.1 => /usr/lib64/libwx_gtk2u_aui-suse.so.1 (0x00007fcf28d06000)
        libwx_gtk2u_core-suse.so.1 => /usr/lib64/libwx_gtk2u_core-suse.so.1 (0x00007fcf28578000)
        libwx_baseu-suse.so.1 => /usr/lib64/libwx_baseu-suse.so.1 (0x00007fcf28133000)
        libwx_gtk2u_adv-suse.so.1 => /usr/lib64/libwx_gtk2u_adv-suse.so.1 (0x00007fcf27d5b000)
        libwx_gtk2u_html-suse.so.1 => /usr/lib64/libwx_gtk2u_html-suse.so.1 (0x00007fcf27a88000)
        libwx_baseu_net-suse.so.1 => /usr/lib64/libwx_baseu_net-suse.so.1 (0x00007fcf27844000)
        libindi.so.1 => /usr/lib64/libindi.so.1 (0x00007fcf27617000)
        libz.so.1 => /lib64/libz.so.1 (0x00007fcf27400000)
        libnova-0.15.so.0 => /usr/lib64/libnova-0.15.so.0 (0x00007fcf26e85000)
        libm.so.6 => /lib64/libm.so.6 (0x00007fcf26b88000)
        libstdc++.so.6 => /usr/lib64/libstdc++.so.6 (0x00007fcf267fe000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007fcf265e7000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007fcf263ca000)
        libc.so.6 => /lib64/libc.so.6 (0x00007fcf26026000)
        libxcb.so.1 => /usr/lib64/libxcb.so.1 (0x00007fcf25e06000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007fcf25c02000)
        libgtk-x11-2.0.so.0 => /usr/lib64/libgtk-x11-2.0.so.0 (0x00007fcf255c8000)
        libgdk-x11-2.0.so.0 => /usr/lib64/libgdk-x11-2.0.so.0 (0x00007fcf25313000)
        libgobject-2.0.so.0 => /usr/lib64/libgobject-2.0.so.0 (0x00007fcf250c2000)
        libpangocairo-1.0.so.0 => /usr/lib64/libpangocairo-1.0.so.0 (0x00007fcf24eb4000)
        libcairo.so.2 => /usr/lib64/libcairo.so.2 (0x00007fcf24b92000)
        libgdk_pixbuf-2.0.so.0 => /usr/lib64/libgdk_pixbuf-2.0.so.0 (0x00007fcf2496f000)
        libpango-1.0.so.0 => /usr/lib64/libpango-1.0.so.0 (0x00007fcf24723000)
        libglib-2.0.so.0 => /usr/lib64/libglib-2.0.so.0 (0x00007fcf24413000)
        libXxf86vm.so.1 => /usr/lib64/libXxf86vm.so.1 (0x00007fcf2420d000)
        libSM.so.6 => /usr/lib64/libSM.so.6 (0x00007fcf24004000)
        libpng16.so.16 => /usr/lib64/libpng16.so.16 (0x00007fcf23dc7000)
        libjpeg.so.8 => /usr/lib64/libjpeg.so.8 (0x00007fcf23b5d000)
        libtiff.so.5 => /usr/lib64/libtiff.so.5 (0x00007fcf238e7000)
        /lib64/ld-linux-x86-64.so.2 (0x0000555ba1467000)
        libnotify.so.4 => /usr/lib64/libnotify.so.4 (0x00007fcf236df000)
        libmspack.so.0 => /usr/lib64/libmspack.so.0 (0x00007fcf234ca000)
        libcfitsio.so.5 => /usr/lib64/libcfitsio.so.5 (0x00007fcf230f2000)
        libXau.so.6 => /usr/lib64/libXau.so.6 (0x00007fcf22eed000)                                                                                                                           
        libgmodule-2.0.so.0 => /usr/lib64/libgmodule-2.0.so.0 (0x00007fcf22ce9000)                                                                                                           
        libXfixes.so.3 => /usr/lib64/libXfixes.so.3 (0x00007fcf22ae3000)                                                                                                                     
        libatk-1.0.so.0 => /usr/lib64/libatk-1.0.so.0 (0x00007fcf228bc000)                                                                                                                   
        libgio-2.0.so.0 => /usr/lib64/libgio-2.0.so.0 (0x00007fcf22535000)                                                                                                                   
        libpangoft2-1.0.so.0 => /usr/lib64/libpangoft2-1.0.so.0 (0x00007fcf2231f000)                                                                                                         
        libfontconfig.so.1 => /usr/lib64/libfontconfig.so.1 (0x00007fcf220e0000)                                                                                                             
        libXrender.so.1 => /usr/lib64/libXrender.so.1 (0x00007fcf21ed6000)                                                                                                                   
        libXinerama.so.1 => /usr/lib64/libXinerama.so.1 (0x00007fcf21cd3000)                                                                                                                 
        libXi.so.6 => /usr/lib64/libXi.so.6 (0x00007fcf21ac2000)                                                                                                                             
        libXrandr.so.2 => /usr/lib64/libXrandr.so.2 (0x00007fcf218b7000)                                                                                                                     
        libXcursor.so.1 => /usr/lib64/libXcursor.so.1 (0x00007fcf216ac000)                                                                                                                   
        libXcomposite.so.1 => /usr/lib64/libXcomposite.so.1 (0x00007fcf214a8000)                                                                                                             
        libXdamage.so.1 => /usr/lib64/libXdamage.so.1 (0x00007fcf212a5000)
        libXext.so.6 => /usr/lib64/libXext.so.6 (0x00007fcf21093000)
        libffi.so.4 => /usr/lib64/libffi.so.4 (0x00007fcf20e89000)
        libfreetype.so.6 => /usr/lib64/libfreetype.so.6 (0x00007fcf20bec000)
        libpixman-1.so.0 => /usr/lib64/libpixman-1.so.0 (0x00007fcf20945000)
        libEGL.so.1 => /usr/X11R6/lib64/libEGL.so.1 (0x00007fcf2073f000)
        libxcb-shm.so.0 => /usr/lib64/libxcb-shm.so.0 (0x00007fcf2053c000)
        libxcb-render.so.0 => /usr/lib64/libxcb-render.so.0 (0x00007fcf20332000)
        libGL.so.1 => /usr/X11R6/lib64/libGL.so.1 (0x00007fcf1fff2000)
        librt.so.1 => /lib64/librt.so.1 (0x00007fcf1fdea000)
        libthai.so.0 => /usr/lib64/libthai.so.0 (0x00007fcf1fbe0000)
        libpcre.so.1 => /usr/lib64/libpcre.so.1 (0x00007fcf1f971000)
        libICE.so.6 => /usr/lib64/libICE.so.6 (0x00007fcf1f754000)
        libuuid.so.1 => /usr/lib64/libuuid.so.1 (0x00007fcf1f54e000)
        liblzma.so.5 => /usr/lib64/liblzma.so.5 (0x00007fcf1f325000)
        libjbig.so.2 => /usr/lib64/libjbig.so.2 (0x00007fcf1f119000)
        libselinux.so.1 => /lib64/libselinux.so.1 (0x00007fcf1eef2000)
        libresolv.so.2 => /lib64/libresolv.so.2 (0x00007fcf1ecdb000)
        libharfbuzz.so.0 => /usr/lib64/libharfbuzz.so.0 (0x00007fcf1ea4b000)
        libexpat.so.1 => /usr/lib64/libexpat.so.1 (0x00007fcf1e821000)
        libbz2.so.1 => /usr/lib64/libbz2.so.1 (0x00007fcf1e611000)
        libGLdispatch.so.0 => /usr/X11R6/lib64/libGLdispatch.so.0 (0x00007fcf1e328000)
        libnvidia-tls.so.375.39 => /usr/lib64/tls/libnvidia-tls.so.375.39 (0x00007fcf1e123000)
        libnvidia-glcore.so.375.39 => /usr/lib64/libnvidia-glcore.so.375.39 (0x00007fcf1c350000)
        libdatrie.so.1 => /usr/lib64/libdatrie.so.1 (0x00007fcf1c14a000)
        libgraphite2.so.3 => /usr/lib64/libgraphite2.so.3 (0x00007fcf1bf21000)

DebugLog, GuideLog and screenshot of frozen application attached.

[PHD2_DebugLog_2017-04-25_215050.txt](https://github.com/OpenPHDGuiding/phd2/files/956362/PHD2_DebugLog_2017-04-25_215050.txt)
[PHD2_GuideLog_2017-04-25_215050.txt](https://github.com/OpenPHDGuiding/phd2/files/956365/PHD2_GuideLog_2017-04-25_215050.txt)
[image]

agalasso, 2017-04-25:
that cmake option is not needed to enable PPEC, and I'm not sure what effect it has.

please follow the standard build instructions: https://github.com/OpenPHDGuiding/phd2/wiki/BuildingPHD2OnLinux

if you see any problem with the standard build, please open an new issue

closed 2017-04-25 by agalasso

domokos, 2017-04-25:
Well in this case the Gaussian option does not show up on the Advanced menu.
[image]

agalasso, 2017-04-25:
did you git clone the latest sources and are you using the main branch?

domokos, 2017-04-26:
I downloaded the Windows release and there I saw the same issue. However I now see what the problem was.

The news announcement said that the feature is present in v2.6.3dev3. I checked out v2.6.3. To me vX.Y.Zwhatever (rc1, rc2, etc.) is the tag/version going for X.Y.Z and not following X.Y.Z.

I did not try but this will be the root cause.

Sorry about it.

domokos, 2017-04-26:
I can confirm that it builds and then works fine when checking out master. Sorry again about raising this.

Another confusing thing is that the maintainer forgot to create tags v2.6.3dev2 and v2.6.3dev3. This again made me think that v2.6.3dev3 is/was heading towards v2.6.3 (without actually checking which version I had been using for guiding before :) and this is why it does not appear on the main branch any more. 

The build version contains v2.6.3dev3 but the git tag itself (including v2.6.3dev2) is missing.