OpenPHDGuiding/phd2 #604: Build fails on Fedora

mattiaverga, 2017-05-03:
I'm trying to build phd2 on Fedora Rawhide, but there seems to be some problem detecting libindi:
`cmake ..
-- The C compiler identification is GNU 7.0.1
-- The CXX compiler identification is GNU 7.0.1
-- Check for working C compiler: /usr/lib64/ccache/cc
-- Check for working C compiler: /usr/lib64/ccache/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/lib64/ccache/c++
-- Check for working CXX compiler: /usr/lib64/ccache/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting build type to 'Release' as none was specified.
-- Performing Test HAS_CXX11_FLAG
-- Performing Test HAS_CXX11_FLAG - Success
-- Performing Test HAS_CXX0X_FLAG
-- Performing Test HAS_CXX0X_FLAG - Success
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.3.5") 
-- Untarring cfitsio
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.21
-- Found PythonInterp: /usr/bin/python (found version "2.7.13") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found wxWidgets: -pthread;;;-lwx_gtk3u_aui-3.0;-lwx_gtk3u_core-3.0;-lwx_baseu-3.0;-lwx_gtk3u_adv-3.0;-lwx_gtk3u_html-3.0;-lwx_baseu_net-3.0 (found version "3.0.3") 
-- Checking for module 'libindi>=0.9'
--   Found libindi, version 1.4.1
CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find INDI (missing: INDI_LIBRARIES) (found suitable version
  "1.4.1", minimum required is "0.9")
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:377 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindINDI.cmake:73 (find_package_handle_standard_args)
  thirdparty/thirdparty.cmake:928 (find_package)
  CMakeLists.txt:126 (include)

-- Configuring incomplete, errors occurred!`

pchev, 2017-05-03:
Hi Mattia,

PHD2 2.6.3 do not compile with INDI 1.4.1.
You need to get the most recent code from Github.

Patrick

mattiaverga, 2017-05-04:
Thank you Patrick, I didn't realize that. Master tree is building fine.

closed 2017-05-04 by mattiaverga