OpenPHDGuiding/phd2 #608: phd2 fails MathToolboxTest on Fedora 26 i386

mattiaverga, 2017-05-18:
Trying to build phd2 on different Fedora versions and architectures gives me an error when running tests.
The MathToolBoxTest fails on Fedora 26 i386, it passes on F26 x86_64 and F25 both i386 and x86_64.

The only changes from F25 to F26 are the upgrade of Eigen3 (F26 uses 3.3.3 which is the same as bundled, while F25 uses 3.2.10) and the upgrade of zlib from 1.2.8 to 1.2.11.

[This is the buildlog from F25 i386](https://copr-be.cloud.fedoraproject.org/results/mattia/Astronomy/fedora-25-i386/00553125-phd2/build.log.gz)

[and this the buildlog from F26 i386](https://copr-be.cloud.fedoraproject.org/results/mattia/Astronomy/fedora-26-i386/00553125-phd2/build.log.gz)

raffienficiaud, 2017-05-19:
Hi,

Would it be possible to have the full output of the failing test? It is possible to have that on stdout/stderr by defining the environment variable `
CTEST_OUTPUT_ON_FAILURE=1`

mattiaverga, 2017-05-19:
I've tried locally using bundled libraries (just to exclude that the problem is about using system libraries) and I get the same error.
The full output is the following:
```
+ env CTEST_OUTPUT_ON_FAILURE=1 make test -C i686-redhat-linux-gnu
make: Entering directory '/builddir/build/BUILD/phd2-master/i686-redhat-linux-gnu'
Running tests...
/usr/bin/ctest --force-new-ctest-process 
Test project /builddir/build/BUILD/phd2-master/i686-redhat-linux-gnu
    Start 1: GaussianProcessTest
1/4 Test #1: GaussianProcessTest ..............   Passed    0.24 sec
    Start 2: MathToolboxTest
2/4 Test #2: MathToolboxTest ..................***Failed    0.23 sec
[==========] Running 11 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 11 tests from MathToolsTest
[ RUN      ] MathToolsTest.BoxMullerTest
[       OK ] MathToolsTest.BoxMullerTest (0 ms)
[ RUN      ] MathToolsTest.UniformMeanTest
[       OK ] MathToolsTest.UniformMeanTest (20 ms)
[ RUN      ] MathToolsTest.BoxMullerMeanTest
[       OK ] MathToolsTest.BoxMullerMeanTest (58 ms)
[ RUN      ] MathToolsTest.RandnMeanTest
[       OK ] MathToolsTest.RandnMeanTest (58 ms)
[ RUN      ] MathToolsTest.RandnStdTest
[       OK ] MathToolsTest.RandnStdTest (63 ms)
[ RUN      ] MathToolsTest.IsNaNTest
[       OK ] MathToolsTest.IsNaNTest (0 ms)
[ RUN      ] MathToolsTest.IsInfTest
[       OK ] MathToolsTest.IsInfTest (0 ms)
[ RUN      ] MathToolsTest.FFTTest
[       OK ] MathToolsTest.FFTTest (1 ms)
[ RUN      ] MathToolsTest.SpectrumTest
/builddir/build/BUILD/phd2-master/contributions/MPI_IS_gaussian_process/tests/gaussian_process/math_tools_test.cpp:181: Failure
The difference between amplitudes(i) and expected_amplitudes(i) is 7.6512304929409103, which exceeds eps, where
amplitudes(i) evaluates to 7.6512304929409103,
expected_amplitudes(i) evaluates to 0, and
eps evaluates to 9.9999999999999995e-07.
/builddir/build/BUILD/phd2-master/contributions/MPI_IS_gaussian_process/tests/gaussian_process/math_tools_test.cpp:182: Failure
The difference between frequencies(i) and expected_frequencies(i) is 0.0625, which exceeds eps, where
frequencies(i) evaluates to 0.1875,
expected_frequencies(i) evaluates to 0.25, and
eps evaluates to 9.9999999999999995e-07.
/builddir/build/BUILD/phd2-master/contributions/MPI_IS_gaussian_process/tests/gaussian_process/math_tools_test.cpp:182: Failure
The difference between frequencies(i) and expected_frequencies(i) is 0.125, which exceeds eps, where
frequencies(i) evaluates to 0.25,
expected_frequencies(i) evaluates to 0.375, and
eps evaluates to 9.9999999999999995e-07.
/builddir/build/BUILD/phd2-master/contributions/MPI_IS_gaussian_process/tests/gaussian_process/math_tools_test.cpp:181: Failure
The difference between amplitudes(i) and expected_amplitudes(i) is 11.935501740402811, which exceeds eps, where
amplitudes(i) evaluates to 4.0644982595971886,
expected_amplitudes(i) evaluates to 16, and
eps evaluates to 9.9999999999999995e-07.
/builddir/build/BUILD/phd2-master/contributions/MPI_IS_gaussian_process/tests/gaussian_process/math_tools_test.cpp:182: Failure
The difference between frequencies(i) and expected_frequencies(i) is 0.1875, which exceeds eps, where
frequencies(i) evaluates to 0.3125,
expected_frequencies(i) evaluates to 0.5, and
eps evaluates to 9.9999999999999995e-07.
[  FAILED  ] MathToolsTest.SpectrumTest (0 ms)
[ RUN      ] MathToolsTest.HammingTest
[       OK ] MathToolsTest.HammingTest (0 ms)
[ RUN      ] MathToolsTest.StdTest
[       OK ] MathToolsTest.StdTest (0 ms)
[----------] 11 tests from MathToolsTest (201 ms total)

[----------] Global test environment tear-down
[==========] 11 tests from 1 test case ran. (202 ms total)
[  PASSED  ] 10 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] MathToolsTest.SpectrumTest

 1 FAILED TEST 

    Start 3: GPGuiderTest
3/4 Test #3: GPGuiderTest .....................   Passed    2.62 sec
    Start 4: GuidePerformanceTest
4/4 Test #4: GuidePerformanceTest .............   Passed   90.06 sec

75% tests passed, 1 tests failed out of 4

Total Test time (real) =  93.21 sec

The following tests FAILED:
          2 - MathToolboxTest (Failed)
Errors while running CTest
```

raffienficiaud, 2017-05-20:
I would say it is an Eigen issue due to GCC 7x. Would it be possible to try compiling both versions with the same compiler, `6.3.1` ?

mattiaverga, 2017-05-28:
Sorry for the delay. I can't easily use GCC 6x in Fedora 26 because it's been built upon GCC 7x. I don't even know if it's possible without rebuilding a lot of packages.
I will try to downgrade GCC and all dependencies as soon as I have some free time.

In the meantime I will try to ask on some Eigen3 mailing list or discussion group if someone is willing to take a look into the problem.

Thanks.

mattiaverga, 2017-06-04:
I managed to install a Fedora 26 machine with gcc downgraded to 6.3.1. Building PHD2 with GCC 6x make the test pass, so the problem seems an incompatibility of MathToolsTest and GCC 7x. (I used the last available Eigen3 version which was built with GCC 7x)

mattiaverga, 2019-05-01:
I cannot see this test failing anymore, maybe some change in gcc or eigen had fixed it (tested with Fedora 29, gcc 8.3.1, eigen 3.3.7)

closed 2019-05-01 by mattiaverga