OpenPHDGuiding/phd2 #609: allow connecting Aux mount if ascom/indi mount cannot report scope position

agalasso, 2017-05-22:
There are some ASCOM mount drivers, like the ZWO USBST4, that cannot report scope position. If one of these is selected as the Mount, then we should allow an Aux Mount selection,. In other words, do not assume that all ASCOM mounts can report position.

SquareBoot, 2018-03-17:
Hello,
can anyone fix this issue? I'm getting
`socket: Connection refused
Dispatch command error(-1): INDI: Could not find property ALIGNTELESCOPECOORDS in EQMod Mount`
with my EQMod mount via INDI and I can't connect it. At least, display a message dialog instead of making PHD2 close.
Kind regards,
SquareBoot

pchev, 2018-03-17:
This error as nothing to do with PHD2, you probably have a version mismatch between libindi and the eqmod driver.
Please open another issue if you are interested to help to fix PHD2 closing in this case, I may need the file you use to install INDI to reproduce the problem.

SquareBoot, 2018-03-17:
You are right, I'll open a new issue