OpenPHDGuiding/phd2 #617: Drift Align Circle Should Not Move After "Adjust" Clicked

downsja1, 2017-07-17:
My understanding was that once you click "Adjust" the alignment circle is supposed to remain fixed in place, allowing you to adjust the mount and move the guide star towards its edge. But there seems to be a bug which occurs from time to time, where as you adjust the mount and start moving the star towards the edge of the circle, the track gate re-appears around the guide star and the alignment circle shifts to be centered around the new position of the guide star.  When this happens, the user is then chasing a moving target, as he was trying to move the guide star closer to the edge of the circle, and now the circle just moved, re-centered itself on the guide star.

agalasso, 2017-07-18:
That's right, the alignment circle is supposed to stay in place when you click Adjust, so I'm not sure what would cause the effect you describe.  If you can reproduce the problem, please post your guide log and debug log in the phd2 forum (http://openphdguiding.org/getting-help/).  When it does happen, please take note of the exact time so we know where to look in the log files.

After looking at the logs we can open a new issue here in the issue tracker if necessary.

closed 2017-07-18 by agalasso

downsja1, 2017-07-18:
I mentioned this bug in a cloudy nights post and first reply I got was someone saying he has seen that happen too. Then another person wrote, "yes, the workaround is to use bookmarks so when that happens you still have markers that haven't moved..."  I will try to get you the guide report and bug report and time of occurrence next time it happens.

James 

----- Reply message -----

agalasso, 2017-07-18:
James, I did not mean to imply that it is not happening. I believe you! In order to best proceed:
 a) we need the logs (Guide Log and Debug Log and time of incident -- the more precise the time the better, ideally down to within a few seconds would be great)
 b) we prefer to discuss the issue in the PHD2 forum first before tracking it here in the issue tracker (http://openphdguiding.org/getting-help/)

I would not recommend CloudyNights as a source of support for PHD2. None of the developers are following the discussions over there, yet we monitor the PHD2 forum very closely and prefer to have issues reported there.

Andy