OpenPHDGuiding/phd2 #624: PHD is never disconnecting the telescope with ASCOM

kevinferrare, 2017-08-17:
I can see here that the connected property of the ASCOM driver is never set to false when clicking on disconnect:
https://github.com/OpenPHDGuiding/phd2/blob/master/scope_ascom.cpp#L416

This can cause issues in some drivers:
 - Reconnection will not work because the resource are not released
 - As the driver receives no disconnection signal, mount movement could continue

agalasso, 2017-08-17:
Hi Kevin, PHD2 releases the driver COM object reference when it disconnects from the scope.  An ASCOM driver _must_ handle the case of a client disconnecting without explicitly setting the Connected property to false. If the driver relies on clients setting Connected = false, then the driver is broken... consider the case of an application crashing or being killed with Task manager. In that scenario the COM object reference is released but Disconnected is never set to false.
You can see some discussion about this in the PHD2 forum, like this discussion thread here: https://groups.google.com/d/msg/open-phd-guiding/ZGJF8oKj4SQ/MaJIbJbIBwAJ
Andy

closed 2017-08-17 by agalasso

kevinferrare, 2017-08-17:
Hi @agalasso, and thanks for the clarification.
Before opening this issue, I tried several things, but I was never able to detect within the driver that the object reference is released (Finalize() and Dispose() methods are never called when clicking on disconnect in PHD).
Do you have any clue on how to catch this within the driver?

agalasso, 2017-08-17:
sure, let's take this discussion to email: andy.galasso@gmail.com

bwdev01, 2017-08-18:
Hi Kevin.  I know you've taken this off-forum with Andy, which is good.  If
you've written your own ASCOM driver, I hope you followed one of the two
templates and didn't just roll your own.  If you used the LocalServer
option, the local server code does its own reference-counting because your
code will see only one client and can't know how many clients there are.
When the last client is disconnected, your driver instance will be disposed
by the LocalServer code.   All of this is tied up in a lot of COM-interop
weirdness, which is why I found it best to use the ASCOM template.  Also,
the 'connected' property turns out to be somewhat ambiguous in the ASCOM
spec - does it mean the client is disconnecting from the driver or the
driver from the mount?  Most people assume the former, but there are some
drivers that handle a 'connected=false' op as a trigger to disconnect from
the mount.  For those drivers, it would be a mess for PHD2 to set the
connected property to false because it would blow all the other clients out
of the water.

 

Hope this helps,

Bruce

 

 

  _____  

with ASCOM (#624)

 

it on GitHub, or mute
2Ds78Xzks5sY_nugaJpZM4O5qUT>  the thread.
<https://github.com/notifications/beacon/AMZ5V3IXRtTcUrwJ9VBOQ9htSbqomOR2ks5
sY_nugaJpZM4O5qUT.gif>

kevinferrare, 2017-08-18:
Hi Bruce, thanks for the insights!

Yes, I used the template "ASCOM Device Driver (C#)".
LocalServer did not seem to make sense for me when I created the driver since:
- The hardware in my case only controls the mount hardware.
- There is already the "POTH Hub (ASCOM)" when several client connection to a single driver is needed.

However, I may be misunderstanding the purpose of LocalServer, since there is very few documentation available on this ...

Regarding the "connected" property, the documentation says "Set True to connect to the device hardware. Set False to disconnect from the device hardware.":
http://www.ascom-standards.org/Help/Developer/html/P_ASCOM_DriverAccess_AscomDriver_Connected.htm
So I followed this when developping the driver ... I may miss something, but it didn't seem very ambiguous to me ...

Something that I don't get is whether it is the responsibility of the driver to handle multiple client connections or not, I thought that the POTH Hub was developed to help drivers focus on the task of interfacing with the hardware.

Kind regards,
Kevin

bwdev01, 2017-08-18:
Hi Kevin. See below for additional comments.  I'm going to be off the
network for several days on an eclipse trip, so I will leave you in Andy's
good hands.  

 

  _____  

with ASCOM (#624)

 

I have a somewhat different take on this, one that led me to use the
LocalServer template.  You have to remember there are several generations of
ASCOM drivers out there, some going back more than 10 years.  Even the old
ones can be completely viable and are still in use.  I think most of the
first-gen drivers were single-user implementations, and I think POTH came
along somewhat later as a way for users to get multiple connections without
the ASCOM device driver author having to rewrite everything.  POTH was done
by the ASCOM developer community as a utility.  But that means it's very
much in the middle of what you're doing and you will probably need to look
at its source code to know how it behaves.  I think the preferred way now is
to implement a LocalServer driver unless the use-case precludes the need for
multiple concurrent clients.  The template is easy enough to use and I've
found it works very well.  By the way, it's also pretty common for people to
first implement with the single-client template, get all the device specific
stuff working, then migrate to LocalServer.  That's what I did and it wasn't
very hard.

http://www.ascom-standards.org/Help/Developer/html/P_ASCOM_DriverAccess_Asco
mDriver_Connected.htm

Maybe the documentation has been updated.  Here's the thread I was thinking
about: 

https://groups.yahoo.com/neo/groups/ASCOM-Talk/conversations/messages/34262

As I recall, we got into this because a specific user was complaining that a
mount disconnect in PHD2 was causing a loss of connection for all his other
client apps.  I can't remember what mount and driver, but I recall that we
had a specific problem that drove us to the current implementation.

But as you see now, you still end up having to think about this stuff.  In
any case, as Andy said, the driver must behave correctly when clients simply
exit or are killed.  Here's a very common use case: an imaging app is
connected to a camera and your mount.  The camera malfunctions, the USB
device connection gets fouled up, and the imaging app is hung.  In all
likelihood, the imaging app will either be exited or killed but it won't go
through a clean disconnect of its devices.  So your mount driver will need
to handle that.  I don't know how POTH deals with those things but it would
probably be easy enough to look at the source code and experiment with it.
Or, as I said, you could migrate to a LocalServer implementation.

Good luck,

Bruce

it on GitHub, or mute
t5mWvDOks5sZaLqgaJpZM4O5qUT>  the thread.
<https://github.com/notifications/beacon/AMZ5V2cOSg144CCzS1a5c-GGJPmlT6gBks5
sZaLqgaJpZM4O5qUT.gif>