OpenPHDGuiding/phd2 #626: INDI cannot connect On-camera mount (sometimes)

agalasso, 2017-08-19:
Steps to reproduce:
  select INDI Camera, configure to connect to INDI CCD Simulator
  Select ON-camera for Mount
  click Connect All
 phd2 reports "Mount Connect Failed"
  sometimes it works though, and I'm not sure what sequence of steps make the differencee

pchev, 2017-08-24:
This must be fixed by 05705297a81ec50ac863b419b74796e90f535c69 
Can you check if it work better now, because I cannot reproduce the problem on my system.

agalasso, 2017-08-24:
It was reproducing reliably for me with an INDI server running on my Linux PC, and connecting from a remote Windows system running PHD2.   With the change, it not longer reproduces, so closing as fixed.
Thanks Patrick!
It is unfortunate that the fix requires an arbitrary wait. Is that a limitation of the protocol that there is not a way to know that there are no more properties to come?

closed 2017-08-24 by agalasso

pchev, 2017-08-24:
The problem is because this properties are send only after the device is physically connected to the indi driver, this is normally done when you click the Connect button. Then the server send all the camera properties but there is no signal when the last one is received. This is not a problem for the mandatory properties we check before to set "modal", but for the optional like the pulse guiding it is not possible to know if it will come later or not.

Another option is to always set m_hasGuideOutput for the indi camera, it will return an error at the first guide command if not supported by the camera.

agalasso, 2017-08-25:
I emailed Jasem and CC'd you.  Let's see what he says.