OpenPHDGuiding/phd2 #628: Help fix the build

zwik, 2017-08-25:
Hey guys,

I was looking trough the commits and found that the build keeps failing. When I click on the red cross to see what's going on the server seems to be down. Is this as designed or should I setup my own environment somehow?

Regards,

Marcel

agalasso, 2017-08-25:
Hi Marcel,

The automated builds are run on some internal machines at [MPIIS Teubingen](https://is.tuebingen.mpg.de/) maintained by Raffi.

When we added INDI support for Mac, the automated builds started failing because the Teubingen OSX build machine does not have the required INDI Client lib installed.  Raffi is aware of this issue and plans on fixing it as soon as he gets a chance.

For the rest of us, the builds do not fail as long as the required prerequisites are installed, as described on the wiki: [Info for Developers](https://github.com/OpenPHDGuiding/phd2/wiki#information-for-developers).

Depending on your needs, you should be able to easily setup a build environment for Windows, OSX, or Linux.  If you just need binaries, you can get them at https://openphdguiding.org/

Andy

raffienficiaud, 2017-08-26:
Hi there,

I am sorry for the noise (currently in vacation BTW). It turned out that the integration of the indi library on the build system requires some more time on my side, as libindi requires cftsio which is included on the build system/source distrib if I remember well, and I want to fix this properly.

So stay tuned ...

zwik, 2017-08-26:
No worries, take your time :) . Was trying to help a bit :) .

closed 2017-08-26 by zwik