OpenPHDGuiding/phd2 #634: HTML help is not valid HTML

zwik, 2017-09-15:
The HTML that is used to display the help is not actually validating correctly. Should the HTML be "strict HTML 4.01"? Or can it also just be HTML5 or something else? If I know what it should be, I can start fixing it :) .

bwdev01, 2017-09-15:
Could you provide some examples using the Help/View mechanism in PHD2 where
the rendering is a significant problem?  The original HTML input is coming
from a third-party word processing app so I don't know how you would try to
fix that.
 
Thanks,
Bruce
 
  _____  

 

<https://github.com/OpenPHDGuiding/phd2/issues/634> , or mute
yfpwWgxks5sim_ggaJpZM4PY6ZO>  the thread.
<https://github.com/notifications/beacon/AMZ5V2Jkqdg9vbFFJ3rIn3CKY2KCtOUZks5
sim_ggaJpZM4PY6ZO.gif>

agalasso, 2017-09-15:
Since the HTML is rendered by the [wxWidgets HTML help viewer](http://docs.wxwidgets.org/3.1/overview_html.html), the primary requirement for the HTML is that it is rendered correctly by wxWidgets and that the anchor tags are correct in order to generate the index and table of contents.
Sadly, wxWidgets only handles a subset of HTML (3?), and would definitely not handle HTML5.

If you want to experiment, you can modify the HTML files, build PHD2 according to the instructions on the wiki, and view the changes by opening the help file from the menu in phd2.

zwik, 2017-09-15:
I'm not having any rendering problems at all. I just like to have the HTML correctly formatted according to the HTML specifications. So documentation is done in a Word document and then converted/save as as HTML? Else it's no use to update the HTML documents. Why not do it in HTML directly? Also how are translations done for those HTML documents? I see some languages have the docs translated as well.

I could fix the HTML and still make sure that wxWidgets can render it. Maybe it's an idea to launch a browser with the help instead?

agalasso, 2017-09-18:
> So documentation is done in a Word document and then converted/save as as HTML?

The HTML in the repository is the definitive source. From the looks of it much of the content may have originated from some HTML editor (not sure what), but what you see in the repo really is the definitive source.

> Also how are translations done for those HTML documents? I see some languages have the docs translated as well.

Translators kind enough to do the work provided us with the localized versions of the manual.

> I could fix the HTML and still make sure that wxWidgets can render it. 

That is generous of you to offer, and I do not have any problem with it, provided the changes are compatible with the wxWidgets renderer.  Bruce?

> Maybe it's an idea to launch a browser with the help instead?

I think that switching the built-in help to pop up in a browser window would be much less usable than the wxHtmlHelp component. I don't see what we'd be gaining by doing that. The limitations of wxWidgets somewhat primitive HTML renderer are not an impediment for the simple, static content we need for the user manual.

FWIW, the source HTML is presented on the Web site (https://openphdguiding.org/manual/?section=Introduction.htm), but Dylan has it going through some kind of filter to match the style of the site.  It's also available un-filtered: https://openphdguiding.org/man/

zwik, 2017-09-18:
Alright, once Bruce also gives his permission I'll get started on fixing the HTML :) .

closed 2017-11-28 by bwdev01