OpenPHDGuiding/phd2 #646: Error message 'Cannot connect mount' with Lodestar INDI driver and 'on camera'

fsignoret, 2017-10-16:
Hello,
After upgrading PHD2 to v2.6.4, I could not connect my Lodestar through ST4 anymore.

PHD2 logs say:
21:08:53.568 00.000 140171388824064 Error thrown from /build/phd2-I60zb7/phd2-2.6.4.rev20171009/scope_onboard_st4.cpp:74->Attempt to Connect Onboard ST4 mount when host does not have guide output
21:08:53.568 00.000 140171388824064 Throw from /build/phd2-I60zb7/phd2-2.6.4.rev20171009/gear_dialog.cpp:1211->OnButtonConnectScope: connect failed
21:08:53.568 00.000 140171388824064 Status Line: Échec de connexion de la monture

Compiled v2.6.3dev5, everything is back to normal.

Suspicious commit:
https://github.com/OpenPHDGuiding/phd2/commit/de56f97b27c9dd026c46d6c572027a7655146123

agalasso, 2017-10-16:
Florian, could you upload your full debug log somewhere and provide a link, or attach the full debug log here?
Andy

fsignoret, 2017-10-16:
[PHD2_DebugLog_2017-10-14_214921.txt](https://github.com/OpenPHDGuiding/phd2/files/1389063/PHD2_DebugLog_2017-10-14_214921.txt)

It also says:
21:49:28.626 00.000 139750451894784 ST4HasGuideOutput=0

pchev, 2017-10-16:
Which INDI version do you use?

Also if you open an INDI client, at the camera General Info tab, what is the value next to Interface:

fsignoret, 2017-10-18:
Client: INDI version: 1.5.0+r3475~201710160747~ubuntu17.04.1
Raspberry: version 1.4.1
indi_sx_ccd version 1.12 interface 6

agalasso, 2017-10-20:
Florian,
Would you mind pulling the latest code and see if you still have the problem?
If there is still a problem, please confirm that you have the lodestar camera device selected here:
[image]
Thanks,
Andy

fsignoret, 2017-10-22:
Hello,
Still failing with 2.6.4.rev20171021 retrieved from patrick's ppa.
I confirm the driver shows "SX CCD LodeStar".
Logs attached.
[PHD2_DebugLog_2017-10-22_204214.txt](https://github.com/OpenPHDGuiding/phd2/files/1405225/PHD2_DebugLog_2017-10-22_204214.txt)

Regards,
Florian

fsignoret, 2017-10-22:
Same result if I compile from the master branch.

fsignoret, 2017-10-22:
I have seen that if I go to the INDI options from PHD2, I can read:
indi_sx_ccd Version 1.12 interface 2
Instead of interface 6 in the INDI control panel of Kstars

fsignoret, 2017-10-22:
[image]

agalasso, 2017-10-23:
That is interesting about the interface version number. Could be a clue but I don't know what to make of it.

Could you try applying this patch and running phd2 and then attaching the debug log?
```
cd phd2
patch -p1 < indi_debug_1.patch
make -C tmp
```

[indi_debug_1.patch.zip](https://github.com/OpenPHDGuiding/phd2/files/1405740/indi_debug_1.patch.zip)

fsignoret, 2017-10-23:
I will do that tonight.
I can already tell you that camera_device->getDriverInterface() gives "2" (I already printed that in debug log yesterday).

fsignoret, 2017-10-23:
[PHD2_DebugLog_2017-10-23_214957.txt](https://github.com/OpenPHDGuiding/phd2/files/1408593/PHD2_DebugLog_2017-10-23_214957.txt)

agalasso, 2017-10-24:
thanks for the log.  The problem is in INDI and is specific to the Lodestar INDI driver (though not necesssarily a problem with the lodestar indi driver itself). I opened a conversation with Jasem to see how we can fix it, and I'll update this issue when I hear back from him.

agalasso, 2017-10-24:
Hi Florian,
I just heard back from Jasem and he made a [fix in INDI](https://github.com/indilib/indi/commit/76865255684165ff9bb75e64c005b051dc34df02). Are you able to [build INDI](https://github.com/indilib/indi#building) to test?  The change is on the INDI server side, so no need to rebuild PHD2.
Andy

fsignoret, 2017-10-24:
Hello,
I have run the steps described until section "Build libindi" included.
I am still getting the same result.
What bothers me is why does PHD2 display an interface different from kstars?
Shouldn't it rely on INDI in the same way?
Florian

agalasso, 2017-10-24:
could you post your phd2 debug log from the run that used the updated INDI server?

fsignoret, 2017-10-24:
Some additional finding: if I start the INDI server from the INDI Web Manager myself, connect the camera on PHD2 first, then it works! The interface shown is 6. Then, I go back on Ekos, I connect the devices, and it shows interface 2!

fsignoret, 2017-10-24:
[PHD2_DebugLog_2017-10-24_220921.txt](https://github.com/OpenPHDGuiding/phd2/files/1412352/PHD2_DebugLog_2017-10-24_220921.txt)

fsignoret, 2017-10-24:
Now, connecting PHD2 first:
[PHD2_DebugLog_2017-10-24_221029.txt](https://github.com/OpenPHDGuiding/phd2/files/1412355/PHD2_DebugLog_2017-10-24_221029.txt)

agalasso, 2017-10-24:
> Some additional finding: if I start the INDI server from the INDI Web Manager myself, connect the camera on PHD2 first, then it works! The interface shown is 6. Then, I go back on Ekos, I connect the devices, and it shows interface 2!

perhaps starting the server one way does not get the updated INDI server code that you just built, but is launching the (older) INDI server installed on the system

fsignoret, 2017-10-24:
How can I revert back to the old INDI server code then? It will allow to verify that.
Thanks
Florian

agalasso, 2017-10-24:
> How can I revert back to the old INDI server code then? It will allow to verify that.

go to your indi local source tree and do
```
git pull
git checkout f0684c2ea7cd2b4f459f1c8e5bff126f4c67e2b7
```
then, to go back to normal, do
  `git checkout master`

fsignoret, 2017-10-24:
It seems it does not change anything.
The first program to connect still gets interface 6, the second one gets interface 2.

agalasso, 2017-10-24:
ok. thank you for your patience and help getting to the bottom of this.   At this point since we know the issue is originating in INDI, I think you should open a new issue in the [indi project](https://github.com/indilib/indi) (or post about the issue in the [INDI forum](http://www.indilib.org/forum.html))

closed 2017-10-25 by fsignoret