OpenPHDGuiding/phd2 #648: bind error in staticpa_toolwin.cpp line 197 column 40

justanenduser, 2017-10-18:
I just pulled the latest from GitHub and encountered the following error during make...

/home/gary/Downloads/phd2-master/staticpa_toolwin.cpp: In constructor ‘StaticPaToolWin::StaticPaToolWin()’:
/home/gary/Downloads/phd2-master/staticpa_toolwin.cpp:197:40: error: cannot bind non-const lvalue reference of type ‘PHD_Point&’ to an rvalue of type ‘PHD_Point’
         PHD_Point radec_now = J2000Now(PHD_Point(c_SthStars.at(is).ra2000, c_SthStars.at(is).dec2000));
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/gary/Downloads/phd2-master/staticpa_toolwin.cpp:37:0:
/home/gary/Downloads/phd2-master/staticpa_toolwin.h:176:15: note:   initializing argument 1 of ‘PHD_Point StaticPaToolWin::J2000Now(PHD_Point&)’
     PHD_Point J2000Now(PHD_Point& radec);

I'm running a fully up2date 64 bit Fedora 26, kernel 4.13.5-200 with gcc 7.2.1-2
The cmake command line and its output are as follows... 

[gary@bigboote build]$ cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local ..
-- The C compiler identification is GNU 7.2.1
-- The CXX compiler identification is GNU 7.2.1
-- Check for working C compiler: /usr/lib64/ccache/cc
-- Check for working C compiler: /usr/lib64/ccache/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/lib64/ccache/c++
-- Check for working CXX compiler: /usr/lib64/ccache/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAS_CXX11_FLAG
-- Performing Test HAS_CXX11_FLAG - Success
-- Performing Test HAS_CXX0X_FLAG
-- Performing Test HAS_CXX0X_FLAG - Success
-- PHD2 version '2.6.4' (extracted from source)
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.3.9") 
-- Untarring cfitsio
-- Found CURL: /usr/lib64/libcurl.so (found version "7.53.1") 
-- Found PythonInterp: /usr/bin/python (found version "2.7.13") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found wxWidgets: -pthread;;;-lwx_gtk3u_aui-3.0;-lwx_gtk3u_core-3.0;-lwx_baseu-3.0;-lwx_gtk3u_adv-3.0;-lwx_gtk3u_html-3.0;-lwx_baseu_net-3.0 (found version "3.0.3") 
-- Checking for module 'INDI'
--   Package 'INDI', required by 'virtual:world', not found
-- Found INDI: /usr/local/include/libindi
-- Found INDI Client Library: /usr/local/lib64/libindiclient.a
-- Found ZLIB: /usr/lib64/libz.so (found version "1.2.11") 
-- 'xgettext' program found at '/usr/bin/xgettext'
-- 'msgfmt' program found at '/usr/bin/msgfmt'
-- 'msgmerge' program found at '/usr/bin/msgmerge'
-- [DOC] Creating HTML target for locale en_EN
-- [DOC] Creating translation target for locale ar_LY
-- [DOC] Creating translation target for locale de_DE
-- [DOC] Creating translation target for locale es_ES
-- [DOC] Creating HTML target for locale fr_FR
-- [DOC] Creating translation target for locale fr_FR
-- [DOC] Creating translation target for locale it_IT
-- [DOC] Creating translation target for locale ja_JP
-- [DOC] Creating translation target for locale ko_KR
-- [DOC] Creating translation target for locale nb_NO
-- [DOC] Creating translation target for locale pl_PL
-- [DOC] Creating translation target for locale pt_BR
-- [DOC] Creating translation target for locale ro_RO
-- [DOC] Creating HTML target for locale ru_RU
-- [DOC] Creating translation target for locale ru_RU
-- [DOC] Creating translation target for locale uk_UA
-- [DOC] Creating translation target for locale zh_CN
-- [DOC] Creating HTML target for locale zh_TW
-- [DOC] Creating translation target for locale zh_TW
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gary/Downloads/phd2-master/build

justanenduser, 2017-10-18:
Omitted two additional error lines...

/home/gary/Downloads/phd2-master/staticpa_toolwin.cpp:213:40: error: cannot bind non-const lvalue reference of type ‘PHD_Point&’ to an rvalue of type ‘PHD_Point’
         PHD_Point radec_now = J2000Now(PHD_Point(c_NthStars.at(is).ra2000, c_NthStars.at(is).dec2000));
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/gary/Downloads/phd2-master/staticpa_toolwin.cpp:37:0:
/home/gary/Downloads/phd2-master/staticpa_toolwin.h:176:15: note:   initializing argument 1 of ‘PHD_Point StaticPaToolWin::J2000Now(PHD_Point&)’
     PHD_Point J2000Now(PHD_Point& radec);
               ^~~~~~~~

agalasso, 2017-10-18:
thanks for the report, @justanenduser

commit 73814ac6 (2017-10-18): fix gcc-7 compile errors. Fixes #648
closed 2017-10-18 by agalasso