OpenPHDGuiding/phd2 #650: Error compiling on armhf with ASI support

tiziano0, 2017-10-22:
Linking CXX executable phd2
CMakeFiles/phd2.dir/cam_ZWO.cpp.o: In function `Camera_ZWO::Connect(wxString const&)':
cam_ZWO.cpp:(.text+0x2e70): undefined reference to `ASIInitCamera'
collect2: error: ld returned 1 exit status
CMakeFiles/phd2.dir/build.make:3351: recipe for target 'phd2' failed
make[2]: *** [phd2] Error 1
CMakeFiles/Makefile2:90: recipe for target 'CMakeFiles/phd2.dir/all' failed
make[1]: *** [CMakeFiles/phd2.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

agalasso, 2017-10-23:
could you attach the full cmake output and make output so we can see the whole context?  even better, the output of
`make VERBOSE=1`

tiziano0, 2017-10-23:
Thank you for your support.
The outputs required are attached

[cmake.log](https://github.com/OpenPHDGuiding/phd2/files/1408674/cmake.log)
[make.log](https://github.com/OpenPHDGuiding/phd2/files/1408675/make.log)

agalasso, 2017-10-24:
thanks for the logs.  I can't tell for sure, but it may be that cmake is picking up a version of ASICamera2.so or ASICamera2.a installed on your system rather than the version bundled with phd2.  Could you check to see if either of those files are present on your system outside the phd2 tree?

tiziano0, 2017-10-24:
I did not have other versions of the file on the system, but I solved adding the following path (Bold) manually to the link.txt file generated after cmake:
.....lib/arm-linux-gnueabihf/libz.so -lASICamera2 **../cameras/zwolibs/armv7/libASICamera2.a** ../cameras/qhyccdlibs/linux/armv7/libqhy.a

agalasso, 2017-10-25:
could you try this patch:

[asi_force_bundled.patch.TXT](https://github.com/OpenPHDGuiding/phd2/files/1413169/asi_force_bundled.patch.TXT)

PatchworkBoy, 2018-02-11:
Same error still occurring on RaspberryPi ModelB+... can confirm supplied patch above resolves issue and should be applied to main repo.

closed 2018-02-12 by agalasso