OpenPHDGuiding/phd2 #661: Warnings issued on build

mattiaverga, 2017-11-09:
Here it is the list of warnings I get when building PHD2 on Fedora (I removed the -Wsign-compare warnings that should already be resolved with the latest commits):
[errors.txt](https://github.com/OpenPHDGuiding/phd2/files/1458783/errors.txt)

agalasso, 2017-11-09:
what compiler and compile flags are you using? I'd like to repro the warnings

mattiaverga, 2017-11-10:
The C compiler identification is GNU 7.2.1
CXXFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic'
I've added '-std=c++14'

If you need more info you can find the full build log on https://copr-be.cloud.fedoraproject.org/results/mattia/Astronomy/fedora-rawhide-x86_64/00658287-phd2/build.log.gz

agalasso, 2017-11-10:
Mattia,
I setup a Fedora 28 (Rawhide) VM and followed the [PHD2 Build Instructions](https://github.com/OpenPHDGuiding/phd2/wiki/BuildingPHD2OnLinux) and got zero warnings or errors.
I'm in favor of fixing all warnings and errors with a standard build (i.e. following the build instructions), but fixing things for a non-standard builds is not going to be productive because developers following the standard build instructions are never going to see the warnings and will never know if they have introduced any new warnings.
I'd like to close this issue since it is not reproducible using the standard PHD2 build instructions.
If you would like to pursue this, I would be open to a PR that increases the warning level and fixes the warnings -- for example by enabling -Wunused-variable in the CMake file.

mattiaverga, 2017-11-11:
Ok, I agree. I started looking to warnings because on Fedora we're using system's Eigen3 libraries, and Eigen 3.3.4 throws in a lot of warnings (will be fixed in 3.3.5).
I then wondered if these warnings were related in some way to #608 , but it doesn't seems so.

closed 2017-11-11 by mattiaverga