OpenPHDGuiding/phd2 #666: Linux, INDI, QHY camera: intermittent camera timeouts in PHD2

agalasso, 2017-12-24:
https://groups.google.com/d/topic/open-phd-guiding/bHn3FSQTnxs/discussion

The INDI log shows the camera exposure complete and the "upload" to PHD2 complete, but phd2 still times out.

fmozza, 2017-12-24:
I repeated the exercise, but used the INDI simulator instead of the QHY5L-II-M device, and was not able to reproduce the time out.  

However, I did notice a problem with the exposure times. PHD2 was configured with a 2 second exposure: the INDI log file shows the 2 second request value, but the acutal time is on the order of a second, and the PHD2 log also shows an actual exposure time of less than 2s.  I checked the previous log files and the times also don't match the requested exposure time. 

I set up the connections again to the real camera, this time setting the exposure time within the INDI window to 2s, but it made no difference - the exposure times that INDI server reported were almost exactly one second.

[dbg2.zip](https://github.com/OpenPHDGuiding/phd2/files/1584630/dbg2.zip)

pchev, 2017-12-26:
This difference in exposure time is strange. 
I cannot directly see how this can prevent phd2 to read the image but this may indicate a more serious problem with the INDI QHY driver. 
 
Do you update the INDI driver to the latest version?
Because it look like there was problem with this driver recently:
http://www.indilib.org/forum/ccds-dslrs/2897-driver-indi-qhy-ccd-crashed-on-virtualbox.html#21801

If this persist you must report that to the INDI forum http://www.indilib.org/forum.html