OpenPHDGuiding/phd2 #667: BLC-induced oscillation

agalasso, 2018-01-07:
[image]

[PHD2_DebugLog_2018-01-06_181007.txt](https://github.com/OpenPHDGuiding/phd2/files/1609409/PHD2_DebugLog_2018-01-06_181007.txt)
[PHD2_GuideLog_2018-01-06_181007.txt](https://github.com/OpenPHDGuiding/phd2/files/1609410/PHD2_GuideLog_2018-01-06_181007.txt)

bwdev01, 2018-01-07:
Those guide pulses on the graph represent a sum of the blc + normal guide
correction.  The blc pulse-size was not being adjusted at those two times -
note that the raw south offsets were increasing at those two points so the
summation simply got larger.  So I don't see a problem with the algo.  You
can also see a history of under-corrections, many instances where a
blc+normal adjustment is followed by a normal adjustment in the same
direction.  That's why the blc pulses were generally being adjusted upward
during this time interval, excluding the two points you're looking at.  So I
don't see a problem beyond the fact that the mount is ill-suited for using
backlash comp and may, as you said, have a backlash comp active in the
firmware.
 
Bruce
  _____  

 
 
<https://user-images.githubusercontent.com/6864470/34645500-23c1bec6-f31d-11
e7-87ed-e1ba03c8e623.png> image

<https://github.com/OpenPHDGuiding/phd2/files/1609409/PHD2_DebugLog_2018-01-
<https://github.com/OpenPHDGuiding/phd2/files/1609410/PHD2_GuideLog_2018-01-
06_181007.txt> 
-
<https://github.com/OpenPHDGuiding/phd2/issues/667> , or mute
CzS6PP1ks5tIBjMgaJpZM4RVf0k>  the thread.
<https://github.com/notifications/beacon/AMZ5VzIX00OFsUrY7Wc1Hw_lQ9SX6WTBks5
tIBjMgaJpZM4RVf0k.gif>

agalasso, 2018-01-07:
thanks Bruce, that makes sense.

bwdev01, 2018-02-10:
Well, it was a plausible explanation, but not apparently the right one. :-(  The mount apparently has a large amount of stiction and stored energy return, resulting in instability (see thread).   We should be able to avoid repeatedly increasing the blc amount in the face of this kind of response although actually compensating for it may be impractical.

bwdev01, 2018-05-05:
Backlash comp adjustment re-written in 2.6.4.dev10 to handle stiction-related issues.

closed 2018-05-05 by bwdev01