OpenPHDGuiding/phd2 #673: Adding USB capability of Camera / Host

deix, 2018-01-23:
I'd like to propose small enhancement to the hardware information field considering guiding cam - adding the information about USB capability of Camera / Host. This could be realized with 2 different ways:

- Full description: 
	Guide Camera USB standard: USB3
	Host USB standard (actual transfer): USB2
- Short version (USB Camera / USB Host)
	(USB3 / USB2)
 
For making it more visual attractive, some USB logos could be added:

Logo for information filed:	https://en.wikipedia.org/wiki/USB#/media/File:USB_Icon.svg
USB2:	https://en.wikipedia.org/wiki/USB#/media/File:Certified_Hi-Speed_USB.svg 
USB3:	https://en.wikipedia.org/wiki/File:SuperSpeed_USB.svg 

Best regards
Michal Powalko

agalasso, 2018-01-23:
Did you have any ideas for where in the PHD2 UI the USB info would be displayed?

deix, 2018-01-23:
Hi Andy,

Yes I do, I have just made some preliminary  "MS Paint" simulation of the 3 palcements for those informations:

[image]

Please replace "HUB" with "Host" - some small error in typing, hard to correct in Paint :)

P.S. The "Guide Stats" could also include "Camera Temperature" for not cooled cams, which I also wrote once :) 

Best regards
Michal Powalko

agalasso, 2018-01-23:
Thx for the image.  That is quite a lot of emphasis.  Is the idea that you are trying to make it obvious when a USB3 camera is plugged-in to a USB2 port? If that is the concern, there may be simpler ways to do it.

For guiding purposes, it is not really important; USB2 speed is quite adequate, especially if subframes are being used (recommended.)  Sure, for planetary imaging you would really want that full USB3 bandwidth but for guiding it really does not matter.

deix, 2018-01-24:
Hi Andy,

I completely agree with You that for guiding the USB2 bandwidth is enough and indeed the idea is mostly to make it obvious when a USB3 camera is plugged-in to a USB2 port. Additional added value of this kind of information could have a "hardware - debugging" aspect as You mentioned above: e.g. using one USB3.0 hub on the mount with all of accessories connected to it, easily accessible information about USB speed of guiding cam provided by PHD2 could give a hint, that something is "slowing down" the connection e.g. the USB2 cable has been used.

I'm aware of the programming effort that need be done into integration of that kind of information, but hope that You will find the way to integrate it :)

Best regards
Michal