OpenPHDGuiding/phd2 #674: drift tool: meridian offset reported incorrectly with INDI

agalasso, 2018-01-28:
https://groups.google.com/d/topic/open-phd-guiding/1QAXL-33eEY/discussion

>  I have the scope pointed to the east. Close to the meridian. And I start drift align. Then in the Drift Align window, with the buttons for Drift and Adjust, it shows Scope Pointing info. THere it says Current Meridian offset: +76. Which feels odd. Indi says: AZ: 179 degrees, HA: 00:00:12. 
>
> So why does PHD2 say Meridian offset +76?

pchev, 2018-02-18:
Andy, do you use the sidereal time returned by ScopeINDI::GetCoordinates  to compute this value?

This is because ScopeINDI return 0 if it is not compiled with libnova, except if the telescope is a lx200, eqmod or a few other that return the sidereal time from the telescope.

pchev, 2018-02-18:
I do more testing and yes this is the reason for the error.

I commit a change to thirdparty.cmake to use libnova if it is present in the system, this solve the problem.

Please can you check I make this cmake change at the right place.

agalasso, 2018-02-18:
looks good, thanks

closed 2018-02-18 by agalasso